gnuradio/gnuradio #5006: xmlrpc_client block doesn't accept address/callback parameters

muaddib1984, 2021-08-15:
xmlrpc client block has variables for address and callback that were changed to dtype raw [here](https://github.com/gnuradio/gnuradio/pull/3554#issue-430591489) which worked in 3.8, now in 3.9 the raw dtype doesn't seem to be evalutated correctly. 
[image]
[image]

if the variables are quoted as strings in the block, the flowgraph can be generated, but the parameters as strings breaks the xmlrpc behavior in the generated flowgraph shown here:
`self.xmlrpc_client_0 = ServerProxy('http://"localhost":8080')`
`...`
`self.xmlrpc_client_0."set_freq"(self.freq)

the desired output would be

`self.xmlrpc_client_0 = ServerProxy('http://localhost:8001')`
`...`
`self.xmlrpc_client_0.set_freq(self.freq)`

i'm submitting a pull request for what i think is the best temporary solution, but it requires that the callback and local variable share the same name.

closed 2024-01-29 by marcusmueller

muaddib1984, 2024-01-29:
> https://gist.github.com/marcusmueller/f2b343e1e23e8fc903708c75d2df766d
> repro5006.svg (view on web)
> can't reproduce, seems to be fixed through #5008

Yes, we fixed this. Sorry for the dangling open issue.<end transmission>

marcusmueller, 2024-01-29:
no worries! Was so happy to a) see it fixed and b) knowing it was your first PR :)