gnuradio/gnuradio #5030: make test error

WangLL1, 2021-08-24:
Hi! I'm building gnu radio from source.When I run "make test",I received the following results.
 The following tests FAILED: 	 
 21 - metainfo_test (Failed) 	 
 78 - qa_rotator_cc (Failed)
 I run "test -V -R metainfo_test" results in
21: Test timeout computed to be: 9.99988e+06
21: E - org.gnuradio.grc.metainfo.xml:org.gnuradio.grc:54
21:     Invalid property for 'url' tag: "contact"
21: 
21: I - org.gnuradio.grc.metainfo.xml:~
21:     Unable to find the curl binary. remote URLs can not be checked for validity!
21: 
21: E - org.gnuradio.grc.metainfo.xml:org.gnuradio.grc:53
21:     Invalid property for 'url' tag: "contact"
21: 
21: E - org.gnuradio.grc.metainfo.xml:org.gnuradio.grc:55
21:     Invalid property for 'url' tag: "contact"
21:

I run "test -V -R qa_rotator_cc" results in
78: Test command: /bin/sh "/home/lrx/GNU_Radio/gnuradio/build/gr-blocks/python/blocks/qa_rotator_cc_test.sh"
78: Test timeout computed to be: 9.99988e+06
78: .....F...
78: ======================================================================
78: FAIL: test_phase_inc_update_out_of_range (__main__.qa_rotator_cc)
78: Test phase increment update sent for an out-of-range offset
78: ----------------------------------------------------------------------
78: Traceback (most recent call last):
78:   File "/home/lrx/GNU_Radio/gnuradio/gr-blocks/python/blocks/qa_rotator_cc.py", line 303, in test_phase_inc_update_out_of_range
78:     self._assert_tags([], [])
78:   File "/home/lrx/GNU_Radio/gnuradio/gr-blocks/python/blocks/qa_rotator_cc.py", line 70, in _assert_tags
78:     self.assertEqual(len(tags), len(expected_tags))
78: AssertionError: 1 != 0
78: 
78: ----------------------------------------------------------------------
78: Ran 9 tests in 0.045s
78: 
78: FAILED (failures=1)
78: 
78: ----------------------------------------------------------------------
78: Tag Debug: rot_phase_inc
78: Input Stream: 00
78:   Offset: 25  Source: n/a     Key: rot_phase_inc   Value: 2.92781
78:   Offset: 50  Source: n/a     Key: rot_phase_inc   Value: 0.699572
78:   Offset: 75  Source: n/a     Key: rot_phase_inc   Value: 2.85284
78: ----------------------------------------------------------------------
78: 
78: ----------------------------------------------------------------------
78: Tag Debug: rot_phase_inc
78: Input Stream: 00
78:   Offset: 50  Source: n/a     Key: rot_phase_inc   Value: -1.88587
78:   Offset: 50  Source: n/a     Key: rot_phase_inc   Value: -0.197396
78:   Offset: 50  Source: n/a     Key: rot_phase_inc   Value: -0.572407
78: ----------------------------------------------------------------------
78: 
78: ----------------------------------------------------------------------
78: Tag Debug: rot_phase_inc
78: Input Stream: 00
78:   Offset: 0  Source: n/a     Key: rot_phase_inc   Value: 0.353997
78: ----------------------------------------------------------------------
78: 
78: ----------------------------------------------------------------------
78: Tag Debug: rot_phase_inc
78: Input Stream: 00
78:   Offset: 25  Source: n/a     Key: rot_phase_inc   Value: 1.68074
78:   Offset: 50  Source: n/a     Key: rot_phase_inc   Value: 2.87367
78:   Offset: 75  Source: n/a     Key: rot_phase_inc   Value: 0.231569
78: ----------------------------------------------------------------------
78: 
78: ----------------------------------------------------------------------
78: Tag Debug: rot_phase_inc
78: Input Stream: 00
78:   Offset: 32768  Source: n/a     Key: rot_phase_inc   Value: 0.628319
78: ----------------------------------------------------------------------
78: 
78: ----------------------------------------------------------------------
78: Tag Debug: rot_phase_inc
78: Input Stream: 00
78:   Offset: 50  Source: n/a     Key: rot_phase_inc   Value: 0.397996
78: ----------------------------------------------------------------------
78: 
78: ----------------------------------------------------------------------
78: Tag Debug: rot_phase_inc
78: Input Stream: 00
78:   Offset: 50  Source: n/a     Key: rot_phase_inc   Value: -1.66284
78: ----------------------------------------------------------------------
1/1 Test #78: qa_rotator_cc ....................***Failed    0.31 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.33 sec

The following tests FAILED:
	 78 - qa_rotator_cc (Failed)

gnu radio:v3.9.2
ubuntu:18.0.4

willcode, 2021-08-25:
This is one of the tests that fails intermittently due to randomness and the use of AlmostEqual. What hardware is this on? It shouldn't be failing every time.

WangLL1, 2021-08-25:
I run on a virtual machine which OS is Ubuntu on VMware Workstation.