gnuradio/gnuradio #5032: Fractional Resampler default Phase Shift parameter

duggabe, 2021-08-29:
In GRC version v3.10.0.0git-429-g655e232a, if I leave the 'Phase Shift' parameter blank in a Fractional Resampler, I get the following error:
<pre>
  File "/home/barry/GRdev/speech_proc_4.py", line 183  
    self.mmse_resampler_xx_0 = filter.mmse_resampler_cc(, 48000/samp_rate)  
                                                        ^  
SyntaxError: invalid syntax
</pre>
Also, GRC shows the Phase Shift as '0' even before opening the Parameters.

A test flowgraph is in https://pastebin.com/4cWJNhjL
Note line 111

solomonbstoner, 2021-08-30:
I had this problem with other blocks. Cant remember specifically which ones, but I know its not just the Fractional Resampler block. I narrowed down the problem to line 217 of grc/core/params/param.py.  The value of the parameter  on the flowgraph is set to `0` by default when the value is empty. I will create a PR to highlight the empty parameter in red.
```python
        elif dtype in ('raw', 'complex', 'real', 'float', 'int', 'short', 'byte', 'hex', 'bool'):
            if expr:
                try:
                    if isinstance(expr, str) and self.is_float(expr[:-1]):
                            scale_factor = expr[-1:]
                            if scale_factor in self.scale:
                                expr = str(float(expr[:-1])*self.scale[scale_factor])
                    value = self.parent_flowgraph.evaluate(expr)
                except Exception as e:
                    raise Exception('Value "{}" cannot be evaluated:\n{}'.format(expr, e))
            else:
                value = 0
            if dtype == 'hex':
                value = hex(value)
            elif dtype == 'bool':
                value = bool(value)
            return value
```

dkozel, 2021-08-31:
The default values of blocks should also be functional, unless there's a strong argument that the block is unusable without the user configuring the values specifically.

@solomonbstoner Your fix for blank values is really useful. First in my mind though is making sure that optional values are not made an error by this. Many blocks don't need all values to be specified.

PR #5035 (2021-09-27): grc: Raise error for unfilled parameters
Fixes #5032 .
Issue 5032 results from an unfilled parameter erroneously indicated as
filled with a default value of 0. This PR makes grc indicate an error
when there is an unfilled parameter by returning None.
closed 2021-09-27 by mormj