gnuradio/gnuradio #5041: gr::log :INFO: packet_headerparser_b0 - Detected an invalid packet at item 48

wsx000, 2021-09-03:
when i try to run the example of **rx_ofdm/tx_ofdm** in **gr_digital/examples/ofdm** , it can't received what i transmitted, and reported as following:
```
gr::log :INFO: packet_headerparser_b0 - Detected an invalid packet at item 0
gr::log :INFO: header_payload_demux0 - Parser returned #f
gr::log :INFO: packet_headerparser_b0 - Detected an invalid packet at item 48
gr::log :INFO: header_payload_demux0 - Parser returned #f
gr::log :INFO: packet_headerparser_b0 - Detected an invalid packet at item 96
gr::log :INFO: header_payload_demux0 - Parser returned #f
gr::log :INFO: packet_headerparser_b0 - Detected an invalid packet at item 144
gr::log :INFO: header_payload_demux0 - Parser returned #f
gr::log :INFO: packet_headerparser_b0 - Detected an invalid packet at item 192
gr::log :INFO: header_payload_demux0 - Parser returned #f
gr::log :INFO: packet_headerparser_b0 - Detected an invalid packet at item 240
gr::log :INFO: header_payload_demux0 - Parser returned #f
......
```
do anyone met the same problem? i can't find the solution.

willcode, 2021-09-03:
@wsx000 please head over to chat.gnuradio.org or the mailing list ... the issue tracker is used for bugs and feature requests. If this turns out to be a bug, reopen this issue with more info.

closed 2021-09-03 by willcode

mmmmmaple, 2023-04-25:
did you solve this problem?I met the same problem and I want to know how to solve this problem if you get rid of it

zzy111-1, 2024-01-24:
Did you solve this problem now?I slightly modified ofdm_tx and rx in the official example, added usrp, and slightly modified parameters such as sampling rate and packet length, so this problem occurred. I searched online and still couldn't solve the problem, so I would like to ask you Any solution ideas?

YamalQRA, 2024-11-10:
hi. did anyone solve the problem? my hardware is usrp b200mini-i
grc in zip attached
[test_codec2)ofdm.zip](https://github.com/user-attachments/files/17692836/test_codec2.ofdm.zip)

marcusmueller, 2024-11-10:
I'm locking this thread. People, you're doing wireless communications – a packet containing error is not inherently a sign of a software bug.

And even more so, we already said that if you need help with something, you should head over to chat or mailing list. `packet_headerparser` has significantly been worked on since 2021, so clearly this specific issue should remain closed Thank you for understanding!