gnuradio/gnuradio #5054: Boost minimum version is not enforced

jwmelto, 2021-09-08:
See https://discourse.cmake.org/t/boost-minimum-version-isnt/4055/2

The `include(GrBoost)` needs to be moved after setting the minimum Boost version in order for it to have effect.

This works:

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 6b2ff192c..b170c9220 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -40,14 +40,6 @@ string(TIMESTAMP BUILD_DATE_SHORT "%Y-%m-%d" UTC)
 message(STATUS "Build date set to ${BUILD_DATE}.")
 
 include(GrComponent)
-########################################################################
-# Setup Boost for global use (within this build)
-# Do this before enabling testing support, as it depends
-# on unit_test_framework
-########################################################################
-include(GrBoost)
-GR_REGISTER_COMPONENT("testing-support" ENABLE_TESTING
-         Boost_UNIT_TEST_FRAMEWORK_FOUND )
 
 # Set the version information here
 SET(VERSION_MAJOR 3)
@@ -70,6 +62,15 @@ set(VOLK_MIN_VERSION "2.4.1")         ## first version with CPU features
 set(PYBIND11_MIN_VERSION "2.4")       # pybind11 sets versions like 2.4.dev4, which compares < 2.4.3
 set(GR_THRIFT_MIN_VERSION "0.13")
 
+########################################################################
+# Setup Boost for global use (within this build)
+# Do this before enabling testing support, as it depends
+# on unit_test_framework
+########################################################################
+include(GrBoost)
+GR_REGISTER_COMPONENT("testing-support" ENABLE_TESTING
+         Boost_UNIT_TEST_FRAMEWORK_FOUND )
+
 # Enable generation of compile_commands.json for code completion engines
 set(CMAKE_EXPORT_COMPILE_COMMANDS ON)
```

willcode, 2021-09-08:
@jwmelto I put in a PR for you.

dkozel, 2021-09-11:
@jwmelto Thanks for chasing this down!

closed 2021-09-19 by mbr0wn