gnuradio/gnuradio #5056: Number Sink does not work for int32

willcode, 2021-09-10:
In `number_sink_impl.cc`, types are detected by size, so `int` is treated as `float`:

```
float number_sink_impl::get_item(const void* input_items, int n)
{
    char* inc;
    short* ins;
    float* inf;

    switch (d_itemsize) {
    case (1):
        inc = (char*)input_items;
        return static_cast<float>(inc[n]);
        break;
    case (2):
        ins = (short*)input_items;
        return static_cast<float>(ins[n]);
        break;
    case (4):
        inf = (float*)input_items;
        return static_cast<float>(inf[n]);
        break;
    default:
        throw std::runtime_error("item size not supported");
    }
    return 0;
}
```

As a result, int types are not displayed correctly.

willcode, 2021-09-10:
PR above removes `int` from the yaml. Any opinions on whether int support can be added or whether it's too much trouble given the implementation? If not, then the PR can close this fix.

platise, 2021-09-11:
I do not understand why you need automatic detection, since in GUI the type is clearly chosen, so, why not using the "selected" type?

willcode, 2021-09-11:
Internally, GNU Radio passes around items of a given size, without regard to type. Blocks have to be designed so they are told an explicit type, or they can guess based on item size. Here, the block was designed to be told item size, which makes float/int ambiguous.

The fix would be to pass a type code/enum, rather than an item size, to the C++ block. This is an API change, so we won't be able to fix 3.8 or 3.9. Hopefully short or float can be used wherever needed.

dkozel, 2021-09-11:
Extending the API on master seems reasonable, we can have a second constructor that takes an enum rather than itemsize? I wonder why it was setup that way originally.

marcusmueller, 2024-01-29:
yeah, we're still only caring about the itemsize.

I've written in-depth instructions on how to fix that, please refer to https://lists.gnu.org/archive/html/discuss-gnuradio/2023-12/msg00049.html

StudHamza, 2024-04-04:
Hi @marcusmueller, I started working on this issue recently, In the mailing list you said 
"Then you would actually need to implement that in number_sink_impl.cc like [1]. But for that you need to modify the actual constructor to not take size_t itemsize as argument [2], but item_type_t itemtype!"

The constructor has a `io_signature::make(nconnections, nconnections, itemsize),` function that expects ‘gr::io_signature::make(int&, int&, gr::qtgui::number_sink::item_type_t&)’, if i remove the itemsize agrument, how do i properly call the `io_signature::make` function?

The error i get when bulding is 
`In constructor ‘gr::qtgui::number_sink_impl::number_sink_impl(gr::qtgui::number_sink::item_type_t, float, gr::qtgui::graph_t, int, QWidget*)’:
gnuradio/gr-qtgui/lib/number_sink_impl.cc:55:36: error: no matching function for call to ‘gr::io_signature::make(int&, int&, gr::qtgui::number_sink::item_type_t&)’
   55 |                  io_signature::make(nconnections, nconnections, itemtype),
      |                  ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~`

[1] https://github.com/gnuradio/gnuradio/blob/86af478a88cf0521b023ad2c924c74ea38d9adcf/gr-qtgui/lib/number_sink_impl.cc#L38-L43
[2] https://github.com/gnuradio/gnuradio/blob/86af478a88cf0521b023ad2c924c74ea38d9adcf/gr-qtgui/lib/number_sink_impl.cc#L46
[3] https://github.com/gnuradio/gnuradio/blob/86af478a88cf0521b023ad2c924c74ea38d9adcf/gr-qtgui/lib/number_sink_impl.cc#L48