gnuradio/gnuradio #5067: High CPU usage by ZMQ Rep Msg Sink

spiccinini, 2021-09-14:
Until the first message is received ZMQ Rep Msg Sink uses one full CPU.

Looking to the code It seems to be a busyloop in rep_msg_sink_impl.cc:
```cpp
    void rep_msg_sink_impl::readloop()
    {
      while(!d_finished) {

        // while we have data, wait for query...
        while(!empty_p(pmt::mp("in"))) {
         ...
      }
   }
```
It can be reproduced with the following flowgraph, until the first msg arrives at t=20s the CPU is at 100%

[image]

I am not sure about the fix, maybe a queue.wait() call is needed.

willcode, 2021-09-14:
We should also check the other _msg_source_ and _msg_sink_ blocks for this kind of behavior.

marcusmueller, 2024-01-29:
https://github.com/gnuradio/gnuradio/blob/d7cdf222756a092bdcdc991f776356c0b9bf4578/gr-zeromq/lib/rep_msg_sink_impl.cc#L93

should be fixed now.

closed 2024-01-29 by marcusmueller