gnuradio/gnuradio #5068: gr_modtool newmod: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa0 in position 8: invalid start byte

platise, 2021-09-14:
For release gnuradio 3.8:

```
Traceback (most recent call last):
  File "/usr/bin/gr_modtool", line 30, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gnuradio/modtool/cli/base.py", line 153, in wrapper
    return func(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gnuradio/modtool/cli/newmod.py", line 64, in cli
    run(self)
  File "/usr/lib/python3/dist-packages/gnuradio/modtool/cli/base.py", line 172, in run
    module.run()
  File "/usr/lib/python3/dist-packages/gnuradio/modtool/core/newmod.py", line 95, in run
    s = filetext.read()
  File "/usr/lib/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa0 in position 8: invalid start byte

```

For some reason the templates contained __pycache__ subfolder, so I've added to line newmod.py:91 check to remove pycache subfolder to fix this error, and added logger info to print what was copied.

```
            for filename in files:
                f = os.path.join(root, filename)
                if "__pycache__" in f:
                    continue
                logger.info(f)
                with open(f, 'r') as filetext:

```

willcode, 2021-09-14:
I don't see any `__pycache__` files in the repo, or in the prefix/share directory on an install. What were you doing when this happened? What templates contained a `pycache` subfolder?

mormj, 2021-09-14:
@platise  - Can you please share the method of installation for gnu radio in your case.  Does it happen to be Ubuntu 18.04 with installation from the PPA?  There is an installation artifact in that case that I'm aware of.

mormj, 2021-09-15:
As a workaround you should be able to 

py3clean /usr/share/gnuradio/modtool/templates/gr-newmod

mormj, 2021-09-15:
This issue has been fixed in Ubuntu 20.04 installs

closed 2022-08-27 by platise