gnuradio/gnuradio #5072: hier_block2 __getattr__ calls itself recursively if it doesn't have the _impl attribute

MattMills, 2021-09-18:
So uh... I was trying to debug something and I found this bug. At least I think it's a bug.

In hier_block2 __getattr__ has a check for if not hasattr(self, "_impl") which raises a runtime error, but the runtime error calls self.__class__.__name__ which calls __getattr__ which checks for if not has attr(self, "_impl") which raises a runtime error, but the runtime error calls...

Well you get the point.

MattMills, 2021-09-18:
Obviously I worked around it on my side by moving my logging statement below __init__ but since it seems like a simple fix to eliminate I thought I'd open an issue.

MattMills, 2021-09-18:
Woops I don't think that branch would actually fix it, I'll have to go poke it in python rather than the github interface some time later if you want a PR from me to fix it.

dkozel, 2021-10-02:
Well, that seems like a nice landmine to remove so other folks can't hit it in the future. Thanks for the report. If you could submit a PR that'd be excellent.

PR #5303 (2021-11-04): gr-runtime: Change hier_block2.py __getattr__ to avoid recursive loop in exception (backport to maint-3.8)
gnuradio-runtime: hier_block2.py __getattr__ recursive loop bug fix #5072

(cherry picked from commit 4457eb145b8078614da1d4b64add6282e4ab8d5d)

Backport https://github.com/gnuradio/gnuradio/pull/5153
commit 114aef31 (2021-11-04): gr-runtime: Change hier_block2.py __getattr__ to avoid recursive loop in exception

gnuradio-runtime: hier_block2.py __getattr__ recursive loop bug fix #5072

(cherry picked from commit 4457eb145b8078614da1d4b64add6282e4ab8d5d)