gnuradio/gnuradio #5074: hashlib.md5 doesn't appear to be compatible with FIPS-enabled OS

joshjab, 2021-09-20:
This isn't really a huge issue for most users I imagine. However, I've noticed that I am unable to get GNURadio compile or install on RedHat 8 with STIG (Security Technical Implementation Guide) applied. I don't think it'll affect many users, but I change a few lines in GrPython.cmake and a few others from 'hashlib.md5' to 'hashlib.sha256' to get it working. 

I don't think making this change creates any major move in core functionality. I'm pretty sure it just creates a hash for making unique builds, but it would allow building GNURadio natively for people with 'FIPS Mode' enabled, which essentially prevents any hashing algorithm below sha256 from being run for whatever reason.

marcusmueller, 2021-09-21:
Yes, I had a run-in with Red Hat's security policy, when they tried to remove a useful hashing function (not its usage in any cryptographic context) from SQLite in fedora. It's got not that much to do with security, sadly, but more with corporate checklists.

So, here in GNU Radio, we use MD5 hashes to track, during development, whether a header file has changed to require updating of Python bindings for that. So, if you change that, you also need to recalculate all `BINDTOOL_HEADER_FILE_HASH(…)` in all `gr-*/python/gnuradio/*/bindings/*.cc`. If you don't, the next CMake run should fail. Not impossible, but work that has absolutely no positive effect on security.

joshjab, 2021-09-22:
+1 on the corporate checklists for sure. And unfortunately it's the only reason it happens to be a necessity for my use case. Not allowed to use any other OS without the FIPS mode enabled amongst the other 'STIG's. 

I hope to get an environment working on Debian to verify it doesn't break anything, but I make that change on every box I install for my organization and it seems to build without noticeable issue. Hopefully I can make a PR soon, and I can be a bit more specific about the error I run into.

Flamewires, 2021-09-24:
Building OOTs is also affected. They use md5 in their cmakes which get generated by gr_modtool. +1 for this for sure.

Flamewires, 2021-09-24:
@joshjab I had spoke with you and said I thought there were already issues with this and it was fixed. Apparently that happened in VOLK:

https://github.com/gnuradio/volk/pull/307
https://github.com/gnuradio/volk/issues/238

michael-roe, 2022-01-16:
It sounds like this isn't actually a security issue, as the has function isn't being used for security.

on the other hand, I have some sympathy for people who are trying to do a code audit of multi-million line code bases to make sure they aren't using an obsolete hash function in a case where it matters. (in the past, I have worked for an operating system vendor - who shall be nameless - who has this problem)

would it be that hard to make the hash function a compile time option? With the understanding that if you change it, you're going to need to rebuild a ton of stuff.

michael-roe, 2022-01-16:
We're already doing at a lot of "check if the platform has this weird library" during configure, so checking if md5 is available doesn't seem too big a deal.

marcusmueller, 2024-01-29:
no issue here.

closed 2024-01-29 by marcusmueller