gnuradio/gnuradio #5080: src: "libyaml not available"

warnes, 2021-09-20:
I'm running GNUradio 3.9.0 from Homebrew on MacOS 11.6.

When I launch `gnuradio-companion`, I always see the following message:
```
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
<<< Welcome to GNU Radio Companion 3.9.2.0 >>>

Block paths:
	/usr/local/Cellar/gnuradio/3.9.2.0_2/share/gnuradio/grc/blocks
```

I've done
```brew install libyaml```

and 
```
% pip install pylibyaml
Requirement already satisfied: pylibyaml in /usr/local/Cellar/gnuradio/3.9.2.0_2/libexec/venv/lib/python3.9/site-packages (0.1.0)
```

but the message remains.

This isn't an urgent issue, just an annoyance, since GRC seems to run without problems.

marcusmueller, 2021-09-30:
This is nothing GNU Radio can fix – it looks like the Python `yaml` module that is installed on your system simply came without `libyaml` support.

`pylibyaml` can monkey-patch that in, but it needs to be imported by the user before `import yaml` is done.

Is GRC really slow at startup? Because if it isn't: that's a WARNING, not an ERROR...

mormj, 2022-01-20:
Closing due to issue being external to GR's ability to fix

closed 2022-01-20 by mormj