gnuradio/gnuradio #5087: How to manipulate data with "channel model"？

liumaolin-931, 2021-09-22:
I am testing the "channel model" module, and I have created the following flow diagram:
[image]
I set Taps to 1, the others remain unchanged
I observed input data and output data, input (0,1,2,3,...198,199), output is (3,4,5,6,...192,193,194), And 195~199 and 0~2 of input data were discarded，The first small part and the next small part of the input sequence are discarded,Why does this happen? I can't understand，does anyone know? thanks

willcode, 2021-09-22:
This can be unexpected, but is how GNU Radio works. There is no guarantee that all samples will be flushed before the flowgraph stops. Blocks have "history" and "output multiples", which can cause samples to be unprocessed. Please continute the conversation on chat (I see that you are there already).

closed 2021-09-22 by willcode

liumaolin-931, 2021-09-22:
Okay, thank you very much for your reply, I will return to the chat