gnuradio/gnuradio #5090: grc: c++ cannot generate variable blocks

solomonbstoner, 2021-09-23:
Variable blocks cannot be generated in C++. They can be generated in Python. The following error is thrown.

> Generate Error: 'value'

I've traced this problem all the way back to v3.8, so I dont think this is a recent problem. I tested with these two builds:
1. v3.8.3.1-21-gce758703 (Python 3.8.10)
2. v3.10.0.0git-543-g7e73e94e (Python 3.8.10)

The following blocks were tested:
1. Constellation Object
2. RRC Filter Taps
3. High-pass Filter Taps
4. Low-pass Filter Taps

It comes from the key `value:` in the variable blocks' yaml files. e.g.
```yaml
value: ${ firdes.low_pass(gain, samp_rate, cutoff_freq, width, win, beta) }
```
Removing that entry gets rid of the error, but the variable block wont be present in the output cpp or hpp files. 

Alternatively, removing that entry and replacing `var_make` with `make` removes the error and allows the variable block to be generated in the output cpp file, but as a block rather than a variable. E.g. Making this change in High-pass Filter Taps' yaml... 
```diff
parameters:
 -   id: gain
@@ -28,7 +28,6 @@ parameters:
     label: Beta
     dtype: float
     default: '6.76'
-value: ${ firdes.high_pass(gain, samp_rate, cutoff_freq, width, win, beta) }
 
 templates:
     imports: from gnuradio.filter import firdes
@@ -41,7 +40,7 @@ templates:
 cpp_templates:
     includes: ['#include <gnuradio/filter/firdes.h>']
     declarations: 'filter::firdes::sptr ${id};'
-    var_make: |-
+    make: |-
         this->${id} = ${id} = firdes.high_pass(${gain}, ${samp_rate}, ${cutoff_freq},\
         ${width}, ${win}, ${beta});
     link: ['gnuradio::gnuradio-filter']
```
... would produce the variable block in the cpp file as a block.
```cpp
// Blocks:
        this->test = test = firdes.high_pass(1.0, 100000, 5000,500, window.WIN_HAMMING, 6.76);
```

I have a hunch the problem lies with the code that deals with `var_make` and `value` yaml keys for c++ in grc. Could someone advise me where I should start looking to fix that problem?

marcusmueller, 2024-01-31:
Repro: 
[image]

https://gist.github.com/marcusmueller/433b222c15bd06c91f3d618710600417

marcusmueller, 2024-01-31:
Can't reproduce, works now! Closing :)

closed 2024-01-31 by marcusmueller

AsriFox, 2024-02-21:
Can reproduce this bug in 3.10.1.1 (Ubuntu 22.04, official repo) and 3.10.7.0 (Ubuntu 22.04, PPA). `firdes` blocks can be generated, but not 'Constellation Object' nor my own block ['DVB Parameters'](https://github.com/AsriFox/gr-dvbs2rx/blob/8571743635ce9ed259feb7ca729a9777ed28e261/grc/dvbs2rx_dvb_params.block.yml). Probably because filter taps are saved as `std::vector`, while others are custom classes.

Example: [cpp_variable.grc](https://github.com/gnuradio/gnuradio/files/14358411/cpp_variable.grc.txt).txt

Log:
```
Generating: '/home/asrifox/Documents/gr-dvbs2rx/build/untitled'
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.
Generate Error: 'value'
>>> Failure
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/gui/Application.py", line 737, in _handle_action
    generator.write()
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/generator/cpp_top_block.py", line 86, in write
    self._variable_types()
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/generator/cpp_top_block.py", line 310, in _variable_types
    if var.params['value'].dtype != 'raw':
KeyError: 'value'
```

Also: how to obtain verbose logs / traceback from `grcc`?