gnuradio/gnuradio #5095: Middle click menu on QT graph widgets difficult to find

Flamewires, 2021-09-24:
Followup from GRC workshop at GRCon, was suggested to put this here for further discussion/comments:

It would be good if this middle click menu could be exposed in some way that was easier to discover. Many long time gnuradio / GRC users don't even know it exists, and there is a ton of functionality within it.

There are other issues with this menu, as getting middle click functionality on a laptop touchpad is possible, but not very common or understood by many users.

tko, 2023-05-05:
Yeah... I've yet to figure out how to middle-click with mac trackpad (Monterey) or if there's a way to configure a modifier-click to do the same thing.

marcusmueller, 2024-01-29:
this needs to be explained and somehow implied in the uI