gnuradio/gnuradio #5096: GRC: Allow for required parameters

jsallay, 2021-09-24:
I frequently produce flowgraphs that I would like to have a required parameter for.  For example, I may have a filename that I want the user to specify.  All parameters in GRC produce optional parameters with default values, so I have to modify the generated python to make the parameter required.  It would be great if GRC would allow me to do this.

I don't have a strong opinion on how this would be implemented in GRC - as an additional option or as a required parameter block.  I would need to be able to specify the value in GRC, so that I can run the flowgraph, but the generated code would not have a default value.

Some thought should go into variable naming.  Would a required parameter be a positional argument?  Would it have a -- in front of the name?  Generally positional parameters are required and optional parameters are not.  This behavior can be changed in argparse, but could lead to confusion.

dev-DTECH, 2023-03-30:
Hey @mormj, I would like to work on this issue.

mormj, 2023-03-30:
@dev-DTECH - that's great.  This would be a good improvement opportunity for GRC.  It would be good to coordinate the development on the GRC development room on chat.gnuradio.org - https://matrix.to/#/#grc-dev:gnuradio.org