gnuradio/gnuradio #5099: Blocks with raw PMT input values

duggabe, 2021-09-25:
Almost all blocks which have PMT message inputs use the name:value format. For example: https://wiki.gnuradio.org/index.php/QT_GUI_Digital_Number_Control

However, there are two which use raw values instead of the name:value format. These are:
- https://wiki.gnuradio.org/index.php/Mute
- https://wiki.gnuradio.org/index.php/Selector

I recommend that the code accept either name:value pairs or raw input, so that existing code still works and the conventional name:value pairs work as well.

jsallay, 2021-09-25:
The new PMT code is designed to allow for flatbuffers of key , value pairs (dictionaries) or single values, among other types.  I don't think this will be difficult to support, but does need to be kept in mind when we integrate the new PMTs into GNU Radio.

jacobagilbert, 2021-09-26:
Is the idea here that the end solution here is that blocks take pairs directly or dictionaries? I absolutely agree that raw PMT value messages to function ports is not a good pattern, but IMO the desired end state is that message based configuration should be done with dictionaries (and ideally they would accept PDUs directly and access the metadata dict in that case). UHD for example moved to dictionary config some time ago.

jsallay, 2021-09-26:
One thing that is really lacking in the current PMT's is introspection.  I'm hoping to greatly improve upon this with the new implementation.  In general, I agree that passing in some form of structed data is a good thing, but I wouldn't want to force that on every block writer.

Also, I'm hoping that in a variety of cases people will decide to define their own flatbuffers structures rather than using a dictionary.  At the end of the day, there should be encouraged style, but I wouldn't want to bake in a particular structure to the design.

PR #7598 (2025-01-03): gr-blocks: Add PDU support to mute and selector blocks

## Description

 Changed the message handlers of the mute and selector blocks so they can receive pdu messages.
Also added a qa test for floats for the mute block and simplified the work function of the mute block
in an attempt to make it more readable.

## Related Issue

Fixes #5099 

## Which blocks/areas does this affect?

The _set-mute_ port of the mute block and the _en_ port of the selector block now 
can receive both raw PMT values and PDUs as me
closed 2025-01-03 by willcode