gnuradio/gnuradio #5101: Permission error using File Sink on Windows 10

SantaTitular, 2021-09-26:
Hello, 

I just installed GNURadio 3.8 x64 on my Windows 10 Pro-based PC. 

I have the following flowgraph where I implement a simple binary signal generator.

[image]

Yet, when I run, I can't seem to save the bytes using the File Sink block

[image]

I tried giving the full access to Gnu Radio 3.8 folders and still can't work around the problem. Anyone can help?

Thanks in advance for the help!

dkozel, 2021-09-26:
Hi @SantaTitular  Two ideas:

Try using a full file path in the File Sink, so `C:\Users\tomas\out_1.dat` for instance.

Two, I see you have some spaces in the current file path and also a symbol in `1 semestre`. Possibly in 3.8 that is not supported. Can you try a folder without spaces and without any special characters?

I recommend joining the [GNU Radio Chat](https://app.element.io/#/room/#gnuradio:gnuradio.org) where folks can help more interactively.

SantaTitular, 2021-09-27:
Hi @dkozel thank you so much for the quick reply. I tested it and indeed your first suggestion worked! By inputting C:\Users\tomas\out_1.dat and C:\Users\tomas\out_2.dat I was able to create/overwrite the files with the bytes.

Regarding your second sugestion, I tested changed the directory of the script towards C:\Users\tomas and just having out1.dat and out2.dat, and still gave me a permission error. Maybe it is a bug?

Thanks, I'll be joining the GNU Radio chat at once!

dkozel, 2021-09-27:
I'm glad that worked. Yes, this sounds like a bad behavior in the block or environment. Would you be willing to do two additional checks to try to narrow down the issues?

1) A path with a space. ie `C:\Users\tomas\test folder\out_1.dat`
2) A path with a special character. ie `C:\Users\tomas\1°Semestre\out_1.dat`

Can you also say where you got the Windows installer and the exact version of GNU Radio. You can find that in GRC under the help menu.

Thanks!

SantaTitular, 2021-09-27:
Hi again @dkozel ,

I tested both of your suggestions and seems that, with the second test, i.e., a path with the special character, a directory error occurred whereas the first test, with a space, caused no issue. It seems that the special character introduced faulty characters in the path (image below)

[image]

I downloaded the windows installer from here: [http://www.gcndevelopment.com/gnuradio/index.htm](http://www.gcndevelopment.com/gnuradio/index.htm) and the exact version is: GR 3.8.2.0 py3.9 

[image]

In sum, it appears that the permission errors occurs when I do not have a specific path for the file (which is solved by placing the path) and path with special characters lead to errors in the path when executing (so no special characters)

dkozel, 2021-09-27:
Perfect, thank you. Possibly 3.9 is better, but this gives us a lot of information about the issue.

dkozel, 2021-09-29:
Duplicate of https://github.com/gnuradio/gnuradio/issues/4323

closed 2021-09-29 by dkozel