gnuradio/gnuradio #5102: Transmitting waveform

AjayaDahal, 2021-09-27:
Hello everyone,
I have been trying to transmit a 5G IQ baseband waveform(complex numbers) generated using MatLab from GNUradio. Can someone please guide me if this is possible? I used file source to load the file and UHD sink to transmit but the spectrum does not match, I am guessing this is because it is transmitting the file instead of a waveform. I tried looking online but couldn’t find much information. I really appreciate your time and help in this matter. 
Thank you.

dkozel, 2021-09-27:
Hi @AjayaDahal That is possible, but you're correct that the file format is likely part of the issue. We'd also have to see your flowgraph to be able to identify if there are any other possible problems.

The GitHub issues are kept for tracking code problems and aren't great for giving advice or help. Can you please join the GNU Radio chat and ask there? There are more people available to answer questions and also who have broader experience.
https://wiki.gnuradio.org/index.php/Chat

closed 2021-09-27 by dkozel

AjayaDahal, 2021-09-27:
Hi @dkozel. I really appreciate your prompt response. Will do.
Thank you.