gnuradio/gnuradio #5104: USRP N321 Lo Sharing

Findoutthg, 2021-09-28:
Hi everyone ,

Since several weeks i am trying to activate Lo sharing on my USRP N321 by the means of gnuradio 3.8.3.1 and uhd driver 4.0.2.

I think the uhd_sink/source block has been well settled, i mean set lo source to external and enable Lo exportation on channel number one. But when I launch the flowgraph, there is no signal on Lo splitter output port then the assigned led did not blink. 

Can someone help me, please ?
Best regards
[test.txt](https://github.com/gnuradio/gnuradio/files/7242806/test.txt)

dkozel, 2021-09-28:
@atrnati and @michaelld, do you have a reference to setting up LO sharing on the N320/N321 using gr-uhd?

michaelld, 2021-09-28:
@dkozel I've never done that. It's very possible that gr-uhd provides the interface to do so. I strongly doubt that if there is an interface in gr-uhd that it is passed through into GRC -- except possibly via a special async message. I'd need to check the code to verify, just as anybody else ...

Findoutthg, 2021-09-30:
Hello dkozel and michaelld, thank you for your help in advance. 
 In my previous message, I have joined the test.py file in order to permit people to test the configuration.

dkozel, 2021-09-30:
@Findoutthg Thanks for attaching that. Unfortunately I don't have an N321 to test with. @atrnati and @mbr0wn may be able to find someone with access and capacity to do so.

michaelld, 2021-09-30:
I have the N321 to test with, but it's still a few days away as I reconfigure my lab after moving office spaces. In the meantime, there are 2 primary documents on LO sharing for N320/N321 that @Findoutthg can process if not done already. [First](https://files.ettus.com/manual/page_usrp_n3xx.html#n3xx_rh_lo_chaining) [Second](https://kb.ettus.com/USRP_N320/N321_LO_Distribution) I can't say these are perfect, but they should provide the basics that are needed. Once those are understood with UHD, then moving into GR Python should work similarly because gr-uhd provides access to the same methods in the same basic way.

muaddib1984, 2022-01-18:
@michaelld have you had time to look into this? I'm trying to do the exact same thing @Findoutthg was, but can't as gr-uhd doesn't have a mechanism to enable the lo_distribution board.
@daniestevez just replied to me on the GNURadio mailing list [here](https://lists.gnu.org/archive/html/discuss-gnuradio/2022-01/msg00092.html) and he has patched UHD and gr-uhd to make this work in gr-ATA.
[UHD here](https://github.com/daniestevez/uhd/commit/0a6da1a3fd5839b862cac740ed702923ed21b096) and [gr-uhd here](https://github.com/daniestevez/gnuradio/commit/f9909bade86045f379f83001de27317cc261807f)

daniestevez, 2022-01-18:
By the way the patches originally come from @devnulling, if I remember correctly.

muaddib1984, 2022-02-07:
@mbr0wn @daniestevez 
Tagging Daniel as I think his changes are the best candidates for upstream to both UHD and gr-UHD, provided TX LO is added. 
[Here](https://lists.ettus.com/empathy/thread/XA3CQQTNTFK5ZUWOYXUYZI3OWHNID2PA?hash=XA3CQQTNTFK5ZUWOYXUYZI3OWHNID2PA#XA3CQQTNTFK5ZUWOYXUYZI3OWHNID2PA) is the solution I came up with and [here](https://lists.ettus.com/empathy/thread/YR2BKKDAO4B5ERI32BJMYOUXTOWZK6R7?hash=YR2BKKDAO4B5ERI32BJMYOUXTOWZK6R7#YR2BKKDAO4B5ERI32BJMYOUXTOWZK6R7) is why I think my changes aren't quite sufficient for upstreaming. They work on my setup but we all know how that goes 😀

marcusmueller, 2024-01-29:
@mbr0wn lacking the hardware, I can't try to reproduce this. Do you think this persists? else, rm -f .

daniestevez, 2024-01-29:
All I can say is that we updated to GNU Radio 3.10.8.0 and UHD 4.6.0 at the ATA recently and patching GR and UHD as before is still required (unless I have missed an alternative way to share the LO). The patches I used are in 

https://github.com/daniestevez/gnuradio/tree/lo_sharing_uhd4_gr310
https://github.com/daniestevez/uhd/tree/lo_sharing_uhd4

I guess if needed we could use the ATA testbed to test other solutions, though validating LO sharing requires observing a signal from the sky, as there is not a good way to insert a calibration signal into all the channels.

marcusmueller, 2024-01-29:
OK I'd definitely say that counts as reproduction.