gnuradio/gnuradio #5107: Qt GUI Dial - weird behaviour

bernerus, 2021-09-28:
The Qt GUI Dial default value seems to be the exact value returned when read from other blocks, not the knob position. This works OK if the scale factor is 1.
 
If it is, say 0.01, the knob is initially set to a 1/100th of angle that represents the default value when the scale factor is applied, and the displayed value is also 1/100th of the default value.

Conversely, if the default value is interpreted as the initial knob position, the returned value does not account for the scale factor - until the knob is moved.

It needs to be documented whether the default value refers to the actual value or the knob position and then either divide with the scale factor to set the initial position or apply the scale factor when setting the default value.

madhavpcm, 2021-10-03:
Hello, I would like to contribute to this issue. It would be great if I know where to begin.

dkozel, 2021-10-03:
@madhavpcm Here is the main source code file for the Dial.
https://github.com/gnuradio/gnuradio/blob/master/gr-qtgui/python/qtgui/dialgauge.py

If you are new to GNU Radio this documentation page about the structure of blocks might be useful, though this bug is probably a straightforward math error in the code.
https://wiki.gnuradio.org/index.php/BlocksCodingGuide
https://wiki.gnuradio.org/index.php/Tutorials

@bernerus Could you supply an example flowgraph? And/or some screenshots showing the current incorrect behavior? Thank you!

bernerus, 2021-10-03:
[qt_gui_dial_weirdness.grc.zip](https://github.com/gnuradio/gnuradio/files/7274174/qt_gui_dial_weirdness.grc.zip)

Here is an example flowgraph and two screenshots.
The dial min and max is 0-100, and the scale factor is 0.1 which means that the actual values should be between 0 and 10. The default value is 5. 

When starting, the dial value says 0.5 but the actual value is 5, then click the dial and move it just a tiny bit, the dial value changes to, say 0.70 and the actual value changes from 5 to 700m. Then set the dial to the mid position, the dial says 5.00 and the actual value says 5.0. 
The problem occurs only upon start, but once the dial has been moved, everything is OK.

[image]

[image]

furiousluck, 2021-10-15:
hi , i want to know how to test my code if i have made some changes?

bernerus, 2023-03-30:
Well, the expected behaviour in the example I supplied would be that when starting, the knob would point straight up showing the value 5.00, then if the knob is turned left, the text for the knob should decrease smoothly and if turning right it should increase smoothly, an in all cases the values shown should be the same, value-wise, i.e. 0.9 may be shown as 900m

marcusmueller, 2024-01-31:
Still present, still a bug.

PR #7254 (2024-05-09): gr-gtqui: Fixed Qt GUI Dial - weird behaviour

## Description

In issue #5107 the Qt GUI dial default seems to be the exact value returned, if the default value is interpreted as the initial knob position, the returned value does not account for the scale factor - until the knob is moved.
Check the issue for more explaination

## Related Issue

This PR will fix #5107 
## Which blocks/areas does this affect?

Only the QT GUI Dial is affected. There is no effect on the performance. 
## Testing Done
closed 2024-05-09 by willcode