gnuradio/gnuradio #5112: network: UDP & TCP blocks segfault when stopped & restarted

argilo, 2021-09-29:
While attempting to update Gqrx to use the new gr-network UDP Sink, I noticed that a segfault occurs if the flowgraph is stopped and restarted. (Gqrx does this in many places, for instance when the user changes the demodulator type or presses the Start/Stop button.)

The segfault occurs because the block deallocates all its resources in the `stop()` method, treating it like a destructor:

https://github.com/gnuradio/gnuradio/blob/65366505b3d04422cb9ec93414e1c8e474f5aa08/gr-network/lib/udp_sink_impl.cc#L164-L196

The UDP Source and TCP Sink blocks have the same problem.

To demonstrate the problem using GRC, create a flow graph containing any of these blocks. Manually edit the code, replacing `tb.start()` with `tb.start(); tb.stop(); tb.start()`. Run the modified Python code, and a segmentation fault will occur.

I think the code in these blocks' `stop()` methods should be moved into their destructors.

@ghostop14

marcusmueller, 2021-09-29:
we really need to document what start/stop are supposed to do – this is an easy bug to happen

ghostop14, 2021-09-29:
You'll notice that a lot of my blocks moved what would normally be destructor code to the stop() function.  I started using this pattern after trying to solve why in the gr-clenabled blocks, OpenCL handles were not getting released in the destructor.  Turned out that SWIG made no guarantees that C++ destructors would be called and recommended that cleanup be done from other mechanisms (hence using stop()).

The documentation for the blocks coding guide states "... users should overload the blocks start() and stop() functions. The start() routine is called when the flow graph is started before the work() thread is spawned. The stop() routine is called when the flow graph is stopped after the work thread has been joined and exited. Users should ensure that the state variables of the block are initialized property in the start() routine."

@marcusmueller , @argilo , correct me if I'm wrong but shouldn't flowgraphs call stop() in a safe way based on that?  I've never had the blocks segfault like that in the past.

argilo, 2021-09-30:
> Users should ensure that the state variables of the block are initialized property in the start() routine."

If the network blocks did this, I don't think there would be a problem. But right now they do this in the constructor.

argilo, 2021-09-30:
The issue is that a block is only constructed once, but it can be started and stopped many times before it is finally destroyed. So anything that is torn down in `stop()` needs to be set up in `start()`.

marcusmueller, 2021-09-30:
> The issue is that a block is only constructed once, but it can be started and stopped many times before it is finally destroyed. So anything that is torn down in `stop()` needs to be set up in `start()`.

Bit out of scope here, but I feel it might be a good idea to find the blocks that actually implement `stop` and add (automatedly?) unit tests that check for in-flowgraph reconfigurability (which calls `stop`/`start`).

> Turned out that SWIG made no guarantees that C++ destructors would be called and recommended that cleanup be done from other mechanisms (hence using stop()).

I know that feeling! Python has extremely unpredictable garbage collection, and SWIG with its very indirect object handling (at least the way we did it, with more or less opaque pointers) makes it worse.

ghostop14, 2021-09-30:
Sounds like the fix is to ensure good start/stop pairing.  For NUMA it's probably better to move allocation into start anyway.

argilo, 2021-11-22:
I started working on a fix here:

https://github.com/gnuradio/gnuradio/compare/master...argilo:fix-network-restart-segfault

So far I've fixed the UDP Sink block. The UDP Source and TCP Sink still need to be fixed.

PR #5372 (2021-12-07): network: fix segfaults when TCP & UDP blocks are restarted
# Pull Request Details

## Description
The TCP and UDP blocks segfault if `start()` is called after `stop()`, because `stop()` frees resources that are not re-allocated by `start()`. To fix this, I've moved resource allocation for these blocks from the constructor to `start()`.

## Related Issue
Fixes #5112.
Also fixes the Gqrx crash noted in https://github.com/gqrx-sdr/gqrx/pull/982.

## Which blocks/areas does this affect?
* TCP Sink
* UDP Sink
* UDP Source

## Testing Done
I've added regression tests. All of them fail with segfaults [... cut]