gnuradio/gnuradio #5115: Cannot specify link libraries for target "Python::NumPy" which is not built   by this project.

jackdondy, 2021-09-30:
When I was installing gnuradio v3.8.3.1 and v3.9.2.0. I end up with this bug:
```
-- Compiler Version: cc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- User set python executable /usr/bin/python3.6
CMake Error at cmake/Modules/GrPython.cmake:101 (target_link_libraries):
  Cannot specify link libraries for target "Python::NumPy" which is not built
  by this project.
Call Stack (most recent call first):
  CMakeLists.txt:318 (include)

-- Configuring incomplete, errors occurred!
```
I look into cmake/Modules/GrPython.cmake, here is line 90 to 101:
```
# advanced cache variable that the user should set to override the numpy include dir
set(Python_NumPy_INCLUDE_DIR ${NUMPY_INCLUDE_DIR} CACHE FILEPATH "NumPy include directory")
mark_as_advanced(Python_NumPy_INCLUDE_DIR)
# output used by modern FindPython, duplicate the behavior
set(Python_NumPy_INCLUDE_DIRS ${Python_NumPy_INCLUDE_DIR})

# target for building with NumPy
add_library(Python::NumPy INTERFACE IMPORTED)
set_target_properties(Python::NumPy PROPERTIES
    INTERFACE_INCLUDE_DIRECTORIES "${Python_NumPy_INCLUDE_DIRS}"
)
target_link_libraries(Python::NumPy INTERFACE Python::Module)
```
This lines in GrPython.cmake doesn't exist in gnuradio v3.7, which is successfully installed in my ubuntu18.04.
Some other info:
```
[INFO] Prefix Python version is: 3.6.9
[INFO] PyBOMBS Version 2.3.4
```

marcusmueller, 2021-09-30:
Might be related to CMake policy CMP0016, because that's what's checked around the place where CMake emits that "Cannot specify link libraries" message.

Which version of CMake are you using (assuming you're running in pybombs: `pybombs run cmake --version`)?

jackdondy, 2021-09-30:
cmake version 3.10.2.
Indeed, a bit old, default on ubuntu 18.04 LTS.

jackdondy, 2021-09-30:
`pybombs run cmake --version`
output:
`2.3.4`
is the version of pybombs, weird.
I use  this:
`$cmake --version
cmake version 3.10.2`
@marcusmueller

jackdondy, 2021-10-01:
@marcusmueller , I manually installed cmake3.18.6 and solve this problem, thank you very much!

marcusmueller, 2021-10-01:
You're welcome! But this is still a bug: if it won't compile with some common version of cmake, then the GNU Radio build system should refuse to try with that version :)

closed 2021-12-15 by dkozel