gnuradio/gnuradio #5119: [ERROR] Problem occurred while building package gnuradio: Build failed.

marcinsztajn, 2021-09-30:
## Problem description
Building GNU Radio library using pybombs crashes
## Command
`pybombs prefix init /home/ubuntu/prefix/gnuradio -R gnuradio-default`

## System
```
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.2 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```
## LOG
```
[INFO] Prefix Python version is: 3.8.10
[INFO] PyBOMBS Version 2.3.4
[WARNING] There already is a prefix in `/home/ubuntu/prefix/gnuradio'.
y
[INFO] Prefix Python version is: 3.8.10
[INFO] Prefix Python version is: 3.8.10
[INFO] Installing default packages for prefix...
[INFO] 
  - gnuradio
[INFO] Phase 1: Creating install tree and installing binary packages:
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: gnuradio
[WARNING] Build dir already exists: /home/ubuntu/prefix/gnuradio/src/gnuradio/build
y
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx: In function ‘PyObject* _wrap_deinterleave_make(PyObject*, PyObject*, PyObject*)’:
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:5601:56: error: no match for ‘operator=’ (operand types are ‘SwigValueWrapper<std::shared_ptr<gr::blocks::deinterleave> >’ and ‘gr::blocks::deinterleave::sptr’ {aka ‘boost::shared_ptr<gr::blocks::deinterleave>’})
 5601 |       result = gr::blocks::deinterleave::make(arg1,arg2);
      |                                                        ^
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:32:21: note: candidate: ‘SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const SwigValueWrapper<T>&) [with T = std::shared_ptr<gr::blocks::deinterleave>]’
   32 |   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
      |                     ^~~~~~~~
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:32:58: note:   no known conversion for argument 1 from ‘gr::blocks::deinterleave::sptr’ {aka ‘boost::shared_ptr<gr::blocks::deinterleave>’} to ‘const SwigValueWrapper<std::shared_ptr<gr::blocks::deinterleave> >&’
   32 |   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:36:21: note: candidate: ‘SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const T&) [with T = std::shared_ptr<gr::blocks::deinterleave>]’
   36 |   SwigValueWrapper& operator=(const T& t) { SwigMovePointer tmp(new T(t)); pointer = tmp; return *this; }
      |                     ^~~~~~~~
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:36:40: note:   no known conversion for argument 1 from ‘gr::blocks::deinterleave::sptr’ {aka ‘boost::shared_ptr<gr::blocks::deinterleave>’} to ‘const std::shared_ptr<gr::blocks::deinterleave>&’
   36 |   SwigValueWrapper& operator=(const T& t) { SwigMovePointer tmp(new T(t)); pointer = tmp; return *this; }
      |                               ~~~~~~~~~^
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:5611:129: error: invalid static_cast from type ‘SwigValueWrapper<std::shared_ptr<gr::blocks::deinterleave> >’ to type ‘const sptr&’ {aka ‘const boost::shared_ptr<gr::blocks::deinterleave>&’}
 5611 |   resultobj = SWIG_NewPointerObj((new gr::blocks::deinterleave::sptr(static_cast< const gr::blocks::deinterleave::sptr& >(result))), SWIGTYPE_p_std__shared_ptrT_gr__blocks__deinterleave_t, SWIG_POINTER_OWN |  0 );
      |                                                                                                                                 ^
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:1191:89: note: in definition of macro ‘SWIG_NewPointerObj’
 1191 | #define SWIG_NewPointerObj(ptr, type, flags)            SWIG_Python_NewPointerObj(NULL, ptr, type, flags)
      |                                                                                         ^~~
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx: In function ‘PyObject* _wrap_exponentiate_const_cci_make(PyObject*, PyObject*, PyObject*)’:
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:5687:66: error: no match for ‘operator=’ (operand types are ‘SwigValueWrapper<std::shared_ptr<gr::blocks::exponentiate_const_cci> >’ and ‘gr::blocks::exponentiate_const_cci::sptr’ {aka ‘boost::shared_ptr<gr::blocks::exponentiate_const_cci>’})
 5687 |       result = gr::blocks::exponentiate_const_cci::make(arg1,arg2);
      |                                                                  ^
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:32:21: note: candidate: ‘SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const SwigValueWrapper<T>&) [with T = std::shared_ptr<gr::blocks::exponentiate_const_cci>]’
   32 |   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
      |                     ^~~~~~~~
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:32:58: note:   no known conversion for argument 1 from ‘gr::blocks::exponentiate_const_cci::sptr’ {aka ‘boost::shared_ptr<gr::blocks::exponentiate_const_cci>’} to ‘const SwigValueWrapper<std::shared_ptr<gr::blocks::exponentiate_const_cci> >&’
   32 |   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:36:21: note: candidate: ‘SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const T&) [with T = std::shared_ptr<gr::blocks::exponentiate_const_cci>]’
[... 926 lines cut ...]
   32 |   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
      |                     ^~~~~~~~
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:32:58: note:   no known conversion for argument 1 from ‘gr::blocks::throttle::sptr’ {aka ‘boost::shared_ptr<gr::blocks::throttle>’} to ‘const SwigValueWrapper<std::shared_ptr<gr::blocks::throttle> >&’
   32 |   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:36:21: note: candidate: ‘SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const T&) [with T = std::shared_ptr<gr::blocks::throttle>]’
   36 |   SwigValueWrapper& operator=(const T& t) { SwigMovePointer tmp(new T(t)); pointer = tmp; return *this; }
      |                     ^~~~~~~~
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:36:40: note:   no known conversion for argument 1 from ‘gr::blocks::throttle::sptr’ {aka ‘boost::shared_ptr<gr::blocks::throttle>’} to ‘const std::shared_ptr<gr::blocks::throttle>&’
   36 |   SwigValueWrapper& operator=(const T& t) { SwigMovePointer tmp(new T(t)); pointer = tmp; return *this; }
      |                               ~~~~~~~~~^
/home/ubuntu/prefix/gnuradio/src/gnuradio/build/gr-blocks/swig/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx:79346:121: error: invalid static_cast from type ‘SwigValueWrapper<std::shared_ptr<gr::blocks::throttle> >’ to type ‘const sptr&’ {aka ‘const boost::shared_ptr<gr::blocks::throttle>&’}
[... cut ...]
make[2]: *** [gr-blocks/swig/CMakeFiles/blocks_swig7.dir/build.make:63: gr-blocks/swig/CMakeFiles/blocks_swig7.dir/CMakeFiles/blocks_swig7.dir/blocks_swig7PYTHON_wrap.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:6042: gr-blocks/swig/CMakeFiles/blocks_swig7.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package gnuradio:
Build failed.
[ERROR] Error installing package gnuradio. Aborting.
```

velichkov, 2021-09-30:
Hi @marcinsztajn,

Do you have another gnuradio version installed from the Ubuntu's repository (in `/usr/`) or manually in (`/usr/local/`)?

If yes the you need to remove it, then delete `/home/ubuntu/prefix/gnuradio/src/gnuradio/build/` directory and retry.

marcinsztajn, 2021-09-30:
> Hi @marcinsztajn,
> 
> Do you have another gnuradio version installed from the Ubuntu's repository (in `/usr/`) or manually in (`/usr/local/`)?
> 
> If yes the you need to remove it, then delete `/home/ubuntu/prefix/gnuradio/src/gnuradio/build/` directory and retry.

Hi @velichkov 
No, I do not have. Its pure new Ubuntu.

willcode, 2021-10-01:
Please move help requests to chat.gnuradio.org or the mailing list. The issue tracker is used for bugs and feature requests. Thanks.

closed 2021-10-01 by willcode