gnuradio/gnuradio #5121: Python: OOT modules install Python packages into non-standard location on Ubuntu

schneider42, 2021-09-30:
Setup: I have Ubuntu 20.04 with GNURadio 3.9.2 from the PPA.

If I install an OOT module which has Python blocks, cmake installs the Python package to `/usr/local/lib/python3/dist-packages/<module name>`.

(At least on Ubuntu) Python does not look for packages at this location, though:
`$ python3 -c 'import sys; print(sys.path)'`
`['', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/schneider/.local/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages']`

Note that `/usr/local/lib/python3.8/dist-packages` is in that list, but `/usr/local/lib/python3/dist-packages` is not (`python3.8` vs. `python3`).

This is the cmake code that gr-modtool generates for Python blocks:
```cmake
GR_PYTHON_INSTALL(
    FILES
    __init__.py
    DESTINATION ${GR_PYTHON_DIR}/howto
)
```

It uses `GR_PYTHON_DIR` do determine where to install the packages. `GR_PYTHON_DIR` is determined like this:
https://github.com/gnuradio/gnuradio/blob/65366505b3d04422cb9ec93414e1c8e474f5aa08/cmake/Modules/GrPython.cmake#L140-L150

On my machine this resolves to:
```
$ python3 -c "from distutils import sysconfig; print(sysconfig.get_python_lib(plat_specific=True, prefix=''))"
lib/python3/dist-packages
```
Which I guess is correct (note that `/usr/lib/python3/dist-packages` is in the search path).

cmake seems to put a `/usr/local` in front of this (which I'd say is correct), but the resulting path does not exist and the package can not be imported.

Now I'm unsure: Who is to blame here?
 - Is GNURadio getting a wrong path?
 - Is distutils fooling us?
   - Can we force it to give us a `3.8` instead of a `3`?
 - Is Ubuntu not adding a relevant entry to the search path?

Maybe someone can help...

drmpeg, 2021-09-30:
It was intentionally changed with this commit:

https://github.com/gnuradio/gnuradio/commit/00822e3808509a98f4f8c59736aeb89cec0d7bbb

I'm not sure why.

schneider42, 2021-09-30:
Well, https://github.com/gnuradio/gnuradio/issues/2006 clearly states that. The previous way might have created the right thing for Ubuntu/Debian, but the wrong thing for other distributions.

schneider42, 2021-09-30:
The [Debian wiki](https://wiki.debian.org/Python) says:
> Debian distributions modify upstream Python in a few ways that are important to understand. Of course, where at all possible, we try to minimize deviations from upstream, but here is an enumeration of the changes you might encounter on a Debian system (and derivatives, such as Ubuntu).

>    dist-packages instead of site-packages. Third party Python software installed from Debian packages goes into dist-packages, not site-packages. This is to reduce conflict between the system Python, and any from-source Python build you might install manually. 

To me this seems to imply that things built from source should go to `site-packages` and not `dist-packages`. The only `site-packages` in the default Ubuntu 20.04 search path is `$HOME/.local/lib/python3.8/site-packages`. I guess we do not want to install OOTs into user specific folders?

The only folder under `/usr/local` which is in the search path is `/usr/local/lib/python3.8/dist-packages`. Setting `GR_PYTHON_DIR` to this also yields the desired result for me.

schneider42, 2021-09-30:
`python3 -c 'import site; print(site.getsitepackages()[0])'`:

Ubuntu 20.04: `/usr/local/lib/python3.8/dist-packages`
pipenv under Ubuntu 20.04: `$HOME/.local/share/virtualenvs/foo-Gc-N9v42/lib/python3.8/site-packages`
pybombs under Ubuntu 20.04: `/usr/local/lib/python3.8/dist-packages`

So that doesn't work under pybombs.

schneider42, 2021-10-01:
Potential Ubuntu issue: https://bugs.launchpad.net/ubuntu/+source/python3-stdlib-extensions/+bug/1832215

schneider42, 2021-10-01:
So I guess this is what is going on:

On Ubuntu (and I guess Debian):
 - Packages installed via `sudo apt` go to `/usr/lib/python3/dist-packages`. Note the `python3` part. These are for all installed Python3 versions (e.g. on Ubuntu 18.04 `python3`, `python3.6` and `python3.8` will all have this in their search path).
 - Packages installed via `sudo pip` go to `/usr/local/lib/python3.X/dist-packages`. Note the `python3.X` part. These are only available for a specific Python version. In most cases that will be the same one as `python3`, but it is possible to have multiple installed at the same time.
 - Packages installed via `pip` go to something like `$HOME/.local/lib/python3.X/site-packages`. Also note the `python3.X` part.

At the moment GNURadio's cmake tries to get a `GR_PYTHON_DIR` which should satisfy both, installations via apt (e.g. the PPA) and installations from source (e.g. compiling GNURadio youself, or compiling an OOT after installing gnuradio-dev from the PPA). cmake later on decides what the prefix for this path should be (`/usr/`, `/usr/local`, or any other prefix as specified e.g. by PyBOMBS).

In my opinion that cant work though. You need to specify your prefix when determining `GR_PYTHON_DIR`. Otherwise the Python version inside the path will not be in the correct format. Just going for `python3.X` will also not work as `/usr/lib/python3.X/dist-packages` does not exist.

Further complicating this is the bug mentioned above inside Ubuntu/Debian. Even if we were to specify the prefix when calling `get_python_lib()` it would give us a wrong answer for `/usr/local`.

schneider42, 2021-10-01:
Looks like SoapySDR has a solution for this:
https://github.com/pothosware/SoapySDR/blob/master/python/get_python_lib.py

Quick test on a Ubuntu 18.04 machine:
```
$ python3 get_python_lib.py /
lib/python3.6/site-packages
$ python3 get_python_lib.py /usr
lib/python3/dist-packages
$ python3 get_python_lib.py /usr/local
lib/python3.6/dist-packages
$ python3.8 get_python_lib.py /usr/local
lib/python3.8/dist-packages
$ python3 get_python_lib.py /some/other/path
lib/python3.6/site-packages
$python3.8 get_python_lib.py /some/other/path
lib/python3.8/site-packages
```

dkozel, 2021-10-02:
I've encountered this issue a few times on Ubuntu when doing installs to the system (uncommon for me but frequent for many users). It'd be great to handle this better. Getting it right for all linuxes is unlikely to be possible, but you can (I think?) override it from CMake for distros/packagers with specific intents. If it works for all of our CI targets then we're in good shape in my mind.

schneider42, 2021-10-02:
> I've encountered this issue a few times on Ubuntu when doing installs to the system (uncommon for me but frequent for many users)

I think it is the natural thing to do for many people. It simply has the fewest dependencies and seems obvious if you installed GNURadio via you package manger and want to install an OOT which is not in the repositories of your distribution. I personally prefer PyBOMBS (which is not without its own issues).

> Getting it right for all linuxes is unlikely to be possible, but you can (I think?) override it from CMake for distros/packagers with specific intents.

I'm giving this a try here: https://github.com/muccc/gr-iridium/blob/master/python/CMakeLists.txt#L21

Over there I've got CI builds for:
 - Ubuntu 20.04 + GNURadio 3.8 via apt
 - Ubuntu 20.04 + GNURadio 3.9 via PPA

> If it works for all of our CI targets then we're in good shape in my mind.

The CI workflow of the gnuradio repository does not actually install GNURadio, I doubt it is of help in its current state. I think a basic test would be to install GNURadio into the system and see Python3 can execute a basic flow graph. I'm not sure how well the docker images support that though.

Another question is the license of the `get_python_lib.py` file mentioned above: Boost Software License - Version 1.0 (from SoapySDR).

Is it possible to include that in this repository? It obviously is not linked, but called via system from cmake.

dkozel, 2021-10-02:
The Boost license is compatible with GNU Radio's GPL v3+ license.
https://www.gnu.org/licenses/license-list.en.html#boost

Adding an install and test step to our CI would be lovely. I was thinking in my previous message just of the list of targets as OSs that we should make sure that installation using `make install` does actually give you a functional install.