gnuradio/gnuradio #5122: log4cpp error of GR3.9 on ubuntu18.04

jackdondy, 2021-09-30:
@mormj , I just came across with the same problem you mentioned above and wonder how you deal with it.
I am installing GR 3.9.2.0, end up with:
```
/usr/include/log4cpp/Priority.hh:107:2: error: ISO C++1z does not allow dynamic exception specifications
  throw(std::invalid_argument);
```
and several similar errors.

_Originally posted by @jackdondy in https://github.com/gnuradio/gnuradio/issues/4736#issuecomment-931599424_

jackdondy, 2021-10-01:
I downloaded and successfully installed [liblog4cpp5v5](https://packages.debian.org/buster/liblog4cpp5v5) and [liblog4cpp5-dev](https://packages.debian.org/buster/liblog4cpp5-dev), version 1.1.3-1 from debian. 
The [version 1.1.3-3](https://packages.debian.org/bullseye/liblog4cpp5v5) of liblog4cpp5v5 couldn't be install on my ubuntu18.04.
I tried to build it from source but failed, maybe because I downloaded the [latest version](https://sourceforge.net/projects/log4cpp/files/latest/download).
Whatever, thank you for your clue @drmpeg .

willcode, 2021-10-01:
Please move help requests to chat.gnuradio.org or the mailing list. Thanks.

closed 2021-10-01 by willcode