gnuradio/gnuradio #5128: Allow embedded python blocks to generate the python block class directly in the py flowgraph

jacobagilbert, 2021-10-02:
Currently embedded python blocks generate a temporary file containing the block class. It would be nice to have an option to put this class directly in the python flowgraph GRC generates. This would make that generated python flowgraph file distributable and standalone. 

I suspect there is a good reason this was not done initially, so there are probably use cases this is not appropriate, hence the option to do this.

willcode, 2021-10-02:
Note that the code for an embedded python block is kept in the GRC file. GRC generate a top level file and one file for each embedded python block, each representing a class imported by the main file. Can you clarify whether you are looking for a single python file to be generated instead of several? This does not help much with distribution, but could prevent naming conflicts/collisions where multiple GRCs have embedded python blocks with the same name.

jacobagilbert, 2021-10-02:
Correct, single python file generated.

skoslowski, 2021-10-02:
You could put the modules in a zip file for deployment. See
https://docs.python.org/3/library/zipapp.html

(von unterwegs gesendet)