gnuradio/gnuradio #5133: Python blocks crash if no reference is kept

Sec42, 2021-10-03:
gnuradio crashes if an (embedded) python block is used, but the reference is not kept around.

Example code:
[crash.txt](https://github.com/gnuradio/gnuradio/files/7273373/crash.txt)

This code runs just fine, but if you delete the line 47 (`self.magic=ctest`), it instead crashes with a segmentation fault.

I suspect this is due to the object going out of scope and being garbage-collected.

I think that `connect` should create a reference & stop the block from going out of scope.

Otherwise some warning/error message would be very useful.

willcode, 2021-10-03:
As long as ctest is part of the flowgraph, it's should be referenced, so it shouldn't need an explicit bookkeeping reference. There could be something wrong in the Python wrapper.

Since the code generated by GRC keeps a reference to all blocks used (e.g., `self.blockid = blk()`) in the top block class, this case may not have been tested before.

marcusmueller, 2024-01-31:
Can't reproduce on main/pre-3.11; trying debian 11's GR 3.10.5;

marcusmueller, 2024-01-31:
doesn't crash on debian 11/ 3.10.5, either. Trying on 3.9.2.0 from Fedora 34… doesn't crash, either. So: can't reproduce!

closed 2024-01-31 by marcusmueller