gnuradio/gnuradio #5134: runtime: logger.cc uses std::filesystem in v3.9.3.0

willcode, 2021-10-04:
A C++-17 feature (std::filesystem) is used in logger.cc, where it replaced boost::filesystem. This works on newer C++ compilers even with `-std=c++14`, so it passes CI, but it fails on some older compilers.

mormj, 2021-10-04:
I ran into this with newsched trying to get it running on Ubuntu 18.  The older compilers have std::experimental::filesystem, but it will probably require some compile tests and conditional `#defines` to get it working uniformly

drmpeg, 2021-10-04:
Looks like it slipped through the backport. std::filesystem isn't used in the 3.9.3.0 version of logger.cc, just boost:filesystem. You can just delete the line, ```#include <boost/filesystem.hpp>``` is already in logger.h.

@mormj You can apt-get install gcc 8 on Ubuntu 18.04.

closed 2021-10-04 by willcode