gnuradio/gnuradio #5139: ctrlport unit tests failing

djpeaco, 2021-10-04:
When compiling from source with ctrlport support, some unit tests fail:
```
The following tests FAILED:
         40 - qa_cpp_py_binding (Failed)
         41 - qa_cpp_py_binding_set (Failed)
         42 - qa_ctrlport_probes (Failed)
```
Running verbosely, they all fail because some bindings are missing, for example:
```
40: ERROR: test_001 (__main__.test_cpp_py_binding)
40: ----------------------------------------------------------------------
40: Traceback (most recent call last):
40:   File "/data/gnuradio-3.9.2.0/gr-blocks/python/blocks/qa_cpp_py_binding.py", line 80, in test_001
40:     v1 = gr.RPC_get_string("pyland", "v1", "unit_1_string",
40: AttributeError: module 'gnuradio.gr' has no attribute 'RPC_get_string'
```
and
```
42: ERROR: test_001 (__main__.test_ctrlport_probes)
42: ----------------------------------------------------------------------
42: Traceback (most recent call last):
42:   File "/data/gnuradio-3.9.2.0/gr-blocks/python/blocks/qa_ctrlport_probes.py", line 37, in test_001
42:     self.probe = blocks.ctrlport_probe2_c("samples", "Complex",
42: AttributeError: module 'gnuradio.blocks' has no attribute 'ctrlport_probe2_c'
```
For the latter the source files defining the bindings are [commented out of the CMakefile](https://github.com/gnuradio/gnuradio/blob/6ac6d36be9cab966905a300716b6e0dcf112bcee/gr-blocks/python/blocks/bindings/CMakeLists.txt#L49), so it makes sense that they are failing. Should the corresponding unit tests be commented out as well?

Not sure where RPC_get_string is supposed to be defined, or if there is something wrong with my build causing it to be missing.

Note this is the same set of failed tests reported in !3801, but the error message is different.

This was with version 3.9.2.0.

drmpeg, 2021-10-14:
This is a difficult issue to solve. Some of the constants for ctrlport are created during build time from the file ```gnuradio-runtime/lib/controlport/thrift/gnuradio.thrift```. So static pybind11 bindings can't be created.

drmpeg, 2021-10-14:
In addition to the unit tests failing, the Ctrlport Probe block is broken. This can be seen by loading the ctrlport example flow graph ```gnuradio/gr-blocks/examples/ctrlport/pfb_sync_test.grc```.

PR #6285 (2022-10-31): blocks: Remove ControlPort tests
## Description
Python bindings for ControlPort were not ported to pybind11. As a result, the corresponding tests all fail, and they are skipped in CI. I think it would be best to remove these tests. They can always be re-added if ControlPort Python bindings are added back someday.

## Related Issue
Fixes #5139.

## Which blocks/areas does this affect?
QA tests for ControlPort.

## Checklist
- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significant chang [... cut]
closed 2022-10-31 by mormj