gnuradio/gnuradio #5140: ModToolException: Could not find gr-newmod source dir.

stackprogramer, 2021-10-05:
In Gnuradio-dev 3.9 when I want to create a block I am faced with this error...  how can solve it...

`
gr_modtool newmod tutorial
ModToolException: Could not find gr-newmod source dir.
`

willcode, 2021-10-05:
Please ask for help on chat.gnuradio.org or the mailing list. The issue tracker is used for bug reports and feature requests. Thanks.

mormj, 2021-10-06:
I think this is a real issue - reproduced on a fresh Ubuntu 20.04 docker with GR installed from the PPA

gr_modtool newmod indeed returns that error.

It should find the template from the prefix, but gnuradio-config-info returns
`./`

the conf files in /etc point to the right place. I'm guessing this is a PPA deb file configuration issue - would be curious if the same issue exists under Debian

willcode, 2021-10-06:
OK. For reference, with maint-3.9 and master on F34:
```
$ gr_modtool newmod tutorial
Creating out-of-tree module in ./gr-tutorial...
Done.
Use 'gr_modtool add' to add a new block to this currently empty module.
```

lofaldli, 2022-01-27:
Running this worked for me on Ubuntu 20.04:
```
$ cd /usr/share/gnuradio/modtool/templates/gr-newmod
$ sudo py3clean .
```
Source: https://wiki.gnuradio.org/index.php/InstallingGR#Modtool_on_Ubuntu

closed 2024-01-31 by marcusmueller