gnuradio/gnuradio #5152: Re-implement support for straight connections between blocks

dkozel, 2021-10-07:
Some folks prefer having straight/90 degree lines for the connections between blocks. The math and code for the connection types is still available from GNU Radio 3.7 and could be cleaned up and integrated as an option so users can set a preference for one or the other.

Current curved implementation
https://github.com/gnuradio/gnuradio/blob/fd45d06b8fa4c398ffc5cac53b78e82344be9e95/grc/gui/canvas/connection.py#L103-L124

3.7 straight line implementation
https://github.com/gnuradio/gnuradio/blob/248f50f3460fe5dde2aa5b05b34eeec78e1439ac/grc/gui/Connection.py#L102-L144

mpentler, 2021-10-07:
Since I spoke to @dkozel about this I’ve made a start trying to implement it, starting with a new True/False entry in Config.py to hold the setting and a new if block to check for that setting inside _make_path in connector.py before deciding whether to draw curved or straight lines.

Biggest problem is converting the old block to the new way of accessing the various methods in the class.

I’m doing this with a not huge Python knowledge, so we’ll see how it goes.

mpentler, 2021-10-17:
Getting on fairly okay with this although I could do with a bit of guidance on what various variables in <3.7 became in >3.8.

https://github.com/mpentler/gnuradio/commits/straight-lines

Right now my branch just draws straight lines between the bounding points and the bezier  control points. The main task left is to continue converting the older code, but without being able to understand what the various bits and pieces are I’m going in a bit blind.

Then tidying & refactoring etc.

minhaj6, 2024-02-25:
Has this been implemented in the main release? I am using a precompiled binary by radioconda on one windows machine. Otherwise I'll have to build the other implementation myself i guess ( i am terrible at building stuff on windows :( )

marcusmueller, 2024-02-25:
@minhaj6  no. Unless an issue is closed or you have clear indication we forgot to close it, you'll have to assume it's still open for a reason ;)

dkozel, 2024-02-27:
@minhaj6 This is a Python only change though, so you could directly change the Python file on your Radioconda based install. That isn't a great answer, but if you do and improve the current patch then we would be closer to having an implementation ready to merge.

PR #7292 (2024-05-09): GRC: Allow for angular connections
- **GRC: Angular connection (by choice via Preferences)**
- **GRC: Make connection line width preference-configurable**

## Description

[image]

## Related Issue

Closes #5152

## Which blocks/areas does this affect?

- Qt GRC
- Hopefully a selling point for Qt GRC for those who want to go back to the 3.7 days, optically :D

## Testing Done
closed 2024-05-09 by willcode