gnuradio/gnuradio #5175: qtgui: removal of pyqwidget() in 3.9 broke OOTs

willcode, 2021-10-10:
PR https://github.com/gnuradio/gnuradio/pull/4994 consolidated `pyqwidget()` and `qwidget()`, removing the former. Backported to v3.9 in https://github.com/gnuradio/gnuradio/pull/5042. This broke the API (reported by gr-osmosdr @smunaut @Sec42).

For v3.9, need to add an alias that restores `pyqwidget()`.

For master, TBD. If we do not add an alias to master, then the function needs to be deprecated in v3.9.

dkozel, 2021-10-11:
And documented in the https://wiki.gnuradio.org/index.php/GNU_Radio_3.10_OOT_Module_Porting_Guide page

777arc, 2021-11-22:
Is there a big downside to just adding in the alias to 3.10 as well and calling it done?

Jachimo, 2022-12-12:
On the chance it's relevant, I just ran into this problem after a brand-new install of GNURadio from the `gnuradio/gnuradio-releases` PPA, which got me:

    gnuradio/jammy,now 3.10.4.0-0~gnuradio~jammy-1 amd64 [installed]

And then I installed osmo-sdr from its own "latest" repository, getting me:

    osmo-sdr/jammy,now 0.1.8.effcaa7-7 amd64 [installed]

And rtl-sdr, AFAIK the version in the Ubuntu repos is the latest one, at least that I can find… it doesn't have a PPA with newer versions in the same way that GNURadio and Osmo-SDR do, I guess?  So that was just installed with `sudo apt install rtl-sdr`, which got me:

    rtl-sdr/jammy,now 0.6.0-4 amd64 [installed]

Anyway... this combination doesn't seem to work. If I try running `osmocom_fft -a rtl -v`, I get this:

```Text
Traceback (most recent call last):
  File "/usr/bin/osmocom_fft", line 714, in <module>
    main()
  File "/usr/bin/osmocom_fft", line 690, in main
    tb = app_top_block(qapp.arguments(), "osmocom Spectrum Browser")
  File "/usr/bin/osmocom_fft", line 297, in __init__
    self.scope_win = sip.wrapinstance(self.scope.pyqwidget(), Qt.QWidget)
AttributeError: 'gnuradio.qtgui.qtgui_python.freq_sink_c' object has no attribute 'pyqwidget'. Did you mean: 'qwidget'?
Segmentation fault (core dumped)
```

willcode, 2022-12-12:
I see the same thing in a recent conda release, but not in a build from current source. Not sure exactly what that means.

In the meantime, if you edit /usr/bin/osmocom_fft and replace all occurances of pyqwidget with qwidget, it should work.

willcode, 2023-05-22:
This has been addressed in various PRs. Closing.

closed 2023-05-22 by willcode