gnuradio/gnuradio #5207: modtool: allow modtool to operate from the build directory

mormj, 2021-10-14:
### Feature Description

Would be nice to be able to do modtool operations from the build/ dir

Here is the sequence I always run into:
'''
gr_modtool add myblock
[go and make the block do something]
cd build
make
[get cmake error that bindings are not present]
gr_modtool bind myblock
[get nasty python error that this is not a module]
```

### Feature Urgency

medium (would be nice to have in the near future)

sb424dat, 2024-03-24:
Can you give more information on the errors? 
Are you the modules you are trying to add?
Any information on versions, OS would assist in recreating the issue.
have you tried [https://lists.gnu.org/archive/html/discuss-gnuradio/2013-03/msg00216.html]?

Jay888-DARK, 2026-10-11:
Hi @marcusmueller, I would like to work on this good first issue! Could you please assign it to me?