gnuradio/gnuradio #5281: std::any breaks gr_modtool bind

drmpeg, 2021-10-30:
### What happened?

The commit https://github.com/gnuradio/gnuradio/commit/041f44e809f39d0567f6f191284d555cd6d08de2 breaks gr_modtool bind.

### System Information

OS: Ubuntu 20.04
GR Installation Method: Source

### GNU Radio Version

3.10-git (master)

### Specific Version

tip of master

### Steps to Reproduce the Problem

In an OOT:

gr_modtool bind <module>

### Relevant log output

```shell
e@w6rz:~/xfer/gr-atsc3$ gr_modtool bind modulator_bc
GNU Radio module name identified: atsc3
INFO Parsing source file "/home/re/xfer/gr-atsc3/include/atsc3/modulator_bc.h" ... 
In file included from /home/re/xfer/gr-atsc3/include/atsc3/modulator_bc.h:13:
In file included from /opt/gnuradio-3.10.0git/include/gnuradio/block.h:17:
In file included from /opt/gnuradio-3.10.0git/include/gnuradio/basic_block.h:15:
In file included from /opt/gnuradio-3.10.0git/include/gnuradio/io_signature.h:17:
In file included from /opt/gnuradio-3.10.0git/include/gnuradio/buffer_double_mapped.h:15:
In file included from /opt/gnuradio-3.10.0git/include/gnuradio/buffer.h:18:
In file included from /opt/gnuradio-3.10.0git/include/gnuradio/tags.h:15:
/opt/gnuradio-3.10.0git/include/pmt/pmt.h:677:35: error: no type named 'any' in
      namespace 'std'
PMT_API pmt_t make_any(const std::any& any);
                             ~~~~~^
/opt/gnuradio-3.10.0git/include/pmt/pmt.h:680:14: error: no type named 'any' in
      namespace 'std'
PMT_API std::any any_ref(pmt_t obj);
        ~~~~~^
/opt/gnuradio-3.10.0git/include/pmt/pmt.h:683:44: error: no type named 'any' in
      namespace 'std'
PMT_API void any_set(pmt_t obj, const std::any& any);
                                      ~~~~~^
3 errors generated.
Error occurred while running CASTXML:  status:1
re@w6rz:~/xfer/gr-atsc3$
```
```

drmpeg, 2021-10-30:
@marcusmueller @mormj

mormj, 2021-11-01:
The issue is that the `apt install python3-pygccxml` version (1.9) doesn't have c++17 support.  This showed up later in pygccxml

pygccxml can be installed from pip and currently gets v 2.2.1

mormj, 2021-11-01:
There is one slight code change that i need to check in to configure pygccxml for c++17, and we should also do some version checking

drmpeg, 2021-11-01:
Just FYI, python3-pygccxml is version 1.9.1 on Ubuntu 21.10 also.

drmpeg, 2021-11-01:
Hmm, I think I see a fix for my RISC-V issue in the latest CastXML. I'll have to give it a try.

drmpeg, 2021-11-01:
The latest CastXML does fix my issue. However, it looks like it wasn't a RISC-V issue, but a gcc 11 issue. So when folks start using gcc 11 (which is in Ubuntu 21.10), expect ```gr_modtool bind``` bug reports.

PR #5288 (2021-11-03): modtool: update pygccxml to c++17 std, add pygccxml version check
# Pull Request Details
Updates the call to pygccxml that is invoked when `gr_modtool bind` is called to use -std c++17
Has minimum version check on pygccxml since the tool is now useless anything below 2.0.0 because we use c++17

## Related Issue
Fixes #5281

## Which blocks/areas does this affect?
gr_modtool

## Testing Done
rebuilt the gnuradio codebase with and without pygccxml installed
created an oot module with a block and ran gr_modtool bind

## Checklist

- [x] I have read the [CONTRIBUTING document](https://github.com/g [... cut]
closed 2021-11-03 by mormj