gnuradio/gnuradio #5292: Segfault when binding with virtual base class

ckoehler, 2021-11-02:
### What happened?

Given two C++ classes:

`class Base`

and 

`class Derived : virtual public Base`

with a method on the base:

`void Base::doSomething()`

The bindtool generates bindings that result in a segfault when calling `doSomething()` from `Derived`, at least when there is also a STL map on `Base` that is accessed in `doSomething()`.

I don't have time to look into it more, and in our case it turned out the virtual base class wasn't needed, fixing the issue. But _if_ it's desired to support this, I don't think it currently is.

### System Information

OS: Ubuntu 20.04
GR Installation Method: source

### GNU Radio Version

3.9 (maint-3.9)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Have the above scenario, build, import into Python and call `doSomething()` on an object of type `Derived`.

ckoehler, 2021-11-02:
Somewhat related, looks like the bindtool doesn't handle virtual stuff well at all. Having a class with pure virtual methods still generates bindings for those methods, and then the build fails because those cannot be bound (naturally).

marcusmueller, 2024-01-31:
Yeah, our C++ polymorphism, templates, overload resolution capabilities are limited, due to the complexity of the language and the fact that pygccxml wouldn't inherently solve this easily.

So, hm, a shortcoming, but I think it's kind of excusable that if someone uses polymorphism in inheritance, they need to look at the wrapper.