gnuradio/gnuradio #5329: gr_modtool: bind fails , if clang is not available

dl1ksv, 2021-11-16:
### What happened?

gr_modtool bind <some oot module> fails with

/usr/local/gnuradio-volker/include/gnuradio/block.h:14:10: fatal error: 'memory' file not found
#include \<memory>
if clang++ is not available
although gcc found this file when building gnuradio from scratch.

If clang++ is available , everything works fine.

### System Information

OS: fedora 33, 35
gnuradio build from scratch

Tested with fedora 33 , pygccxml 2.0.1 and fedora 35 python3.10, pygccxml 2.2.1 

### GNU Radio Version

3.10-git (master)

### Specific Version

v3.10.0.0git-632-g04cf420f

### Steps to Reproduce the Problem

For testing I used [gr-funcube](https://github.com/dl1ksv/gr-funcube)

gr_modtool bind fcd_control[

### Relevant log output

```shell
gr_modtool bind fcd
GNU Radio module name identified: funcube
INFO Parsing source file "/home/schroer/gnuradiocomponents/gr-funcube/include/funcube/fcd.h" ... 
In file included from /home/schroer/gnuradiocomponents/gr-funcube/include/funcube/fcd.h:11:
In file included from /usr/local/gnuradio-volker/include/gnuradio/hier_block2.h:15:
In file included from /usr/local/gnuradio-volker/include/gnuradio/basic_block.h:15:
/usr/local/gnuradio-volker/include/gnuradio/io_signature.h:14:10: fatal error: 'functional' file not found
#include <functional>
         ^~~~~~~~~~~~
1 error generated.
Error occurred while running CASTXML xml file does not exist

It seems, that castxml does not use the default includes for gcc.
```

bkerler, 2022-12-13:
I'm getting the same error, even if clang++ (Ubuntu clang version 14.0.0-1ubuntu1) is installed, so far I wasn't able to find any solution so far.

```
gr_modtool bind websocket_pdu
GNU Radio module name identified: ais_simulator
In file included from /tmp/tmpadte_0vq.h:1:
In file included from /home/gnuradio/src/gr-ais_simulator/include/gnuradio/ais_simulator/websocket_pdu.h:25:
/home/gnuradio/include/gnuradio/block.h:14:10: fatal error: 'memory' file not found
#include <memory>
         ^~~~~~~~
1 error generated.
Error occurred while running CASTXML xml file does not exist

```

bkerler, 2022-12-13:
This issue is solved with installing g++ version 12
`sudo apt install g++-12`

dl1ksv, 2022-12-15:
I tested with f36 and gcc-Version 12.2.1 . Without using clang enabled I get
gr_modtool bind fcd
GNU Radio module name identified: funcube

In file included from /tmp/tmprr8cqgwr.h:1:
In file included from /home/schroer/gnuradiocomponents/gr-funcube/include/funcube/fcd.h:11:
In file included from /usr/local/gnuradio-volker/include/gnuradio/hier_block2.h:15:
In file included from /usr/local/gnuradio-volker/include/gnuradio/basic_block.h:15:
/usr/local/gnuradio-volker/include/gnuradio/io_signature.h:17:10: fatal error: 'functional' file not found
#include <functional>
         ^~~~~~~~~~~~
1 error generated.
Error occurred while running CASTXML xml file does not exist.

Enabling clang results in

gr_modtool bind fcd
GNU Radio module name identified: funcube

Writing binding code to python/bindings/fcd_python.cc
Writing binding code to python/bindings/docstrings/fcd_pydoc_template.h

marcusmueller, 2024-01-31:
I think this got solved by distro bugfix releases to g++ and/or by us moving to C++17? Anyways, can't reproduce on F36, nor ubuntu

closed 2024-01-31 by marcusmueller