gnuradio/gnuradio #5337: Errors when using the "python_message_passing.py"

mikotaFull, 2021-11-19:
God morning,

I want to generate a source controlled by voltage. basically, I send messages to a source in order to change its amplitude. 
I have built a block to generate messages and pass them to the source. To do so, I created an "embedded python block " and used  part of the of the " qa_python_message_passing.py " which is in your repository.

When I execute the graph in gnuradio I get the error  :  "File "/home/epy_block_0.py", line 29, in work self.msg_list.append(pmt.from_double(input_items[0][:]))
File "/usr/lib/python3/dist-packages/pmt/pmt_swig.py", line 1985, in from_double return _pmt_swig.from_double(x)
TypeError: in method 'from_double', argument 1 of type 'double'
thread[thread-per-block[4]: <block message generator(2)>]: SWIG director method error. Error detected when calling 'feval_ll.eval'"

I don't really understand what is wrong., why the pmt.from_double method doesn't work if the argument I'm passing is a double?
Please could you give me some guidance ? I am new to gnuradio and python, any help will be appreciated..
I attach to issue the python code of the block (the message generator) and the GRC flowgraph.

[image]
[image]

willcode, 2021-11-19:
This is a problem with your code on line 29 (`input_items[0][:]` is an array not a single value). We use the issue tracker for bug reports and feature requests, so please head over to chat.gnuradio.org or the mailing list for help.

closed 2021-11-19 by willcode

mikotaFull, 2021-11-20:
Good morning @willcode ,
Thank you for the insight! and sorry I didn't knew there was a chat.
Have a good weekend.