gnuradio/gnuradio #5341: Type error in QT GUI Range Widget

ruurdjan, 2021-11-20:
### What happened?

Using the QT GUI Range widget in either Counter, Slider or Counter+Slider mode causes a Type Error, regardless of Widget Type.

### System Information

OS: Fedora Linux 35 (Workstation Edition)
GR Installation Method: native distro packages - 3.9.2.0 (Python 3.10.0)

### GNU Radio Version

3.9 (maint-3.9)

### Specific Version

9.3.2.0

### Steps to Reproduce the Problem

Use a QT GUI Range widget in either Counter, Slider or Counter+Slider mode.

### Relevant log output

```shell
Traceback (most recent call last):
  File "/home/ruurdjan/rangetest.py", line 199, in <module>
    main()
  File "/home/ruurdjan/rangetest.py", line 177, in main
    tb = top_block_cls()
  File "/home/ruurdjan/rangetest.py", line 87, in __init__
    self._freq_win = RangeWidget(self._freq_range, self.set_freq, 'Frequency', "counter_slider", int, QtCore.Qt.Horizontal)
  File "/usr/lib64/python3.10/site-packages/gnuradio/qtgui/range.py", line 121, in __init__
    self.d_widget = self.CounterSlider(self, self.range, self.notifyChanged, rangeType, orientation)
  File "/usr/lib64/python3.10/site-packages/gnuradio/qtgui/range.py", line 262, in __init__
    self.slider = RangeWidget.Slider(parent, ranges, self.sliderChanged, rangeType, orientation)
  File "/usr/lib64/python3.10/site-packages/gnuradio/qtgui/range.py", line 161, in __init__
    self.setRange(0, ranges.nsteps - 1)
TypeError: setRange(self, int, int): argument 2 has unexpected type 'float'
```

willcode, 2021-11-20:
There's already a PR in to fix this ... linked it to this issue. This should be fixed in the next releases.

ruurdjan, 2021-11-20:
Thanks! I couldn't for the life of me find it in the Issue tracker but didn't search the commits.

willcode, 2021-11-20:
Sometimes the fix gets in with no issue. You can do a search like this:

https://github.com/gnuradio/gnuradio/issues?q=is%3Aopen+range+

PR #5287 (2021-11-22): qtgui range: setRange() requires int
Author: Chris Vine <vine35792468@gmail.com>

# Pull Request Details

## Description
Python 3.10 sees setRange( non-int-value ) as an error. Reported by Chris Vine.

## Related Issue
Fixes #5341 

## Which blocks/areas does this affect?
qtgui/range

## Testing Done

## Checklist

- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/master/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significant change per commit. 
- [x] I [have signed my commits before making this PR](https://github.com/gnuradio/gnuradio/b [... cut]
closed 2021-11-22 by mormj