gnuradio/gnuradio #5349: libfmt dependency not checked by CMake

dkozel, 2021-11-24:
### What happened?

I've just tried compiling master on Ubuntu 20.04 and it failed as libfmt was missing. I haven't checked in depth, but wanted to quickly file an issue incase it is a hard requirement, in which case we should add a CMake check.

### System Information

Source build on Ubuntu 20.04 on Windows Subsystem for Linux

### GNU Radio Version

3.10-git (master)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Build master without libfmt installed.

### Relevant log output

```shell
gr310 [dkozel@DESKTOP-QA3QGAR gnuradio]  (master) cmake --build build${PREFIX}                                                                                [27/1608] Building CXX object gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx.gch                                                    FAILED: gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx.gch                                                                          /usr/bin/c++  -DHAVE_COSF -DHAVE_GETPAGESIZE -DHAVE_MMAP -DHAVE_PTHREAD_SETSCHEDPARAM -DHAVE_PTHREAD_SIGMASK -DHAVE_SCHED_SETSCHEDULER -DHAVE_SHM_OPEN -DHAVE_SIGACTION -DHAVE_SIGNAL_H -DHAVE_SINCOS -DHAVE_SINCOSF -DHAVE_SINF -DHAVE_SYSCONF -DHAVE_SYS_IPC_H -DHAVE_SYS_MMAN_H -DHAVE_SYS_SHM_H -DHAVE_SYS_TYPES_H -DSPDLOG_COMPILED_LIB -I../gnuradio-runtime/include -Ignuradio-runtime/include -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -O3 -DNDEBUG   -std=c++17 -Winvalid-pch -x c++-header -include /home/dkozel/src/gnuradio/buildgr310/gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx -MD -MT gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx.gch -MF gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx.gch.d -o gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx.gch -c gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx.cxx                                                                                                                                         In file included from ../gnuradio-runtime/include/gnuradio/custom_lock.h:15,                                                                                                   from ../gnuradio-runtime/include/gnuradio/buffer.h:15,                                                                                                        from ../gnuradio-runtime/include/gnuradio/buffer_double_mapped.h:15,                                                                                          from ../gnuradio-runtime/include/gnuradio/io_signature.h:17,                                                                                                  from ../gnuradio-runtime/include/gnuradio/basic_block.h:15,                                                                                                   from ../gnuradio-runtime/include/gnuradio/hier_block2.h:15,                                                                                                   from /home/dkozel/src/gnuradio/gnuradio-runtime/include/gnuradio/top_block.h:15,                                                                              from /home/dkozel/src/gnuradio/buildgr310/gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx:5,                                         from <command-line>:                                                                                                                         ../gnuradio-runtime/include/gnuradio/logger.h:25:10: fatal error: fmt/core.h: No such file or directory                                                          25 | #include <fmt/core.h>
```

marcusmueller, 2021-11-24:
point is, we use spdlog, not fmtlib. 

SPDLOG's cmake config is a bit of a mess, in that it hardcodes wheter to use the SPDLOG-integrated fmtlib, or to use the system fmtlib (and mixing the two is not a good idea). On different distros, that .config file is different.

In one case, you need external fmtlib, in the other you don't.

This is my current understanding.

We could probably extract the info about whether it uses internal/external fmtlib, and if external, require fmtlib. Probably, we should fix spdlog upstream so that in a couple of years, the consumer can set spdlog's behaviour.

dkozel, 2021-11-24:
Thanks for the details! SPDLog was correctly picked up during CMake generation. Lets make sure the fmtlib error is included in the installation pages of the wiki at least.

marcusmueller, 2021-11-24:
I think it's fair to consider this an issue until we've fixed non-checking for fmtlib in case of external configuration of spdlog; doing it right trumps documenting the wrong behaviour :)

marcusmueller, 2021-11-24:
@dkozel could you check whether this works with fmt installed and complains without?
```
diff --git a/gnuradio-runtime/CMakeLists.txt b/gnuradio-runtime/CMakeLists.txt
index a81a5c142..03ea483cc 100644
--- a/gnuradio-runtime/CMakeLists.txt
+++ b/gnuradio-runtime/CMakeLists.txt
@@ -23,6 +23,12 @@ GR_CHECK_LINUX_SCHED_AVAIL()
 # Find spdlog
 ########################################################################
 find_package(spdlog CONFIG)
+if(SPDLOG_FMT_EXTERNAL)
+  find_package(fmt CONFIG)
+  set(fmt_found_or_internal ${fmt_FOUND})
+else()
+  set(fmt_found_or_internal ON)
+endif()

 ########################################################################
 # Register component
@@ -33,6 +39,7 @@ GR_REGISTER_COMPONENT("gnuradio-runtime" ENABLE_GNURADIO_RUNTIME
     PYTHONINTERP_FOUND
     MPLIB_FOUND
     spdlog_FOUND
+    fmt_found_or_internal
     pybind11_FOUND
 )
```

dkozel, 2021-11-25:
I couldn't make it fail after removing libfmt-dev. Additionally if spdlog itself is missing this falls through and sets `fmt_found_or_internal` to ON.

marcusmueller, 2021-11-25:
'doh. yeah, that logical bug... should've seen that.

michael-roe, 2021-11-25:
I just ran into this issue, too.

ryanvolz, 2021-12-03:
Adding a link here, since it's related to checking `SPDLOG_FMT_EXTERNAL`: https://github.com/gnuradio/gnuradio/pull/4981#issuecomment-952359418. Implementing that should be able to fix the conda builds with the way spdlog is packaged there.

jboone, 2021-12-08:
I got bit by this too.

For what it's worth, new `conda` dependencies `spdlog` and `fmt` aren't in the CondaInstall Wiki entry:
https://wiki.gnuradio.org/index.php/CondaInstall#Create_a_development_environment

I suppose I should request a Wiki account and fix it myself...

ryanvolz, 2021-12-08:
> Edit: I added spdlog and fmt dependencies for Linux to the CondaInstall Wiki page. I have no idea if Windows and MacOS require them as well, so left those alone. Their dependency lists are surprisingly different from Linux.

I added them for Windows and macOS since their use is universal.

I've been trying to dig into this because I want to make a conda build of the release candidate, and it would also be great to have the conda CI working again. I'm finding that now that I've gotten conda to use spdlog 1.9 instead of 1.8.5, it builds fine as long as fmt is installed. Because [logger.h includes the fmt headers directly](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/include/gnuradio/logger.h#L36-L37), I think fmt needs to be installed to build gnuradio-runtime regardless of the value of `SPDLOG_FMT_EXTERNAL`.

However, I think we could avoid this bug and drop the fmt dependency for conda (where spdlog is built with `SPDLOG_FMT_EXTERNAL=OFF`) by changing those fmt imports to
```
#include <spdlog/fmt/fmt.h>
```
which is a [helper header](https://github.com/gabime/spdlog/blob/ff80d10820f1f2482f1fe8a7928849ecfbde5a52/include/spdlog/fmt/fmt.h) that will include the appropriate internal/external fmt headers. I'm also not sure of the point of [the lines here to force defining SPDLOG_FMT_EXTERNAL](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/include/gnuradio/logger.h#L41-L43), particularly in the context of switching to that helper header, so maybe those can be dropped too. These changes appear to be working for me locally in a conda build without an installed external fmt package, but I'm not sure if there are any implications I've missed (particularly for building against gnuradio-runtime with OOTs).

@marcusmueller any idea if what I'm proposing should work, or if there's a reason not to use that helper header?

PR #5380 (2021-12-10): runtime: Use <spdlog/fmt/fmt.h> header handling SPDLOG_FMT_EXTERNAL
Regardless of how spdlog was built (with/without `SPDLOG_FMT_EXTERNAL`), this ensures that the proper fmt headers are included. This makes it so that fmt is not a required dependency when spdlog is built with `SPDLOG_FMT_EXTERNAL=OFF`.

Fixes #5349. Gets the conda CI running against spdlog.
closed 2021-12-10 by mormj