gnuradio/gnuradio #5351: pygccxml 2.0 dependency

michael-roe, 2021-11-25:
### What happened?

So this is probably something that is annoying, rather than a bug.

I am building gnuradio from source on Ubuntu 20.04.

The build seems to fail because I do not have a sufficiently recent version of pygccxml.
The version that comes with this Ubuntu distribution is 1.9.1:
  $ pip install pygccxml
  Requirement already satisfied: pygccxml in /usr/lib/python3/dist-packages (1.9.1)

But it looks like I need version 2.0.

Anyone got any suggestions for how to resolve this?

### System Information

OS: (Linux Distro, Mac, Windows)
GR Installation Method: (native distro packages - apt, rpm, etc., Conda, Source

### GNU Radio Version

3.10-git (master)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3 $(GNURADIO_SRC)

### Relevant log output

```shell
- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3.6.5") 
-- Python checking for numpy - found
-- Python checking for pygccxml - not found
-- 
-- ***************************** WARNING!!! *************************
-- pygccxml is highly recommended for using gr_modtool
-- and is either not present or below the minimum version 2.0.0
-- Only trivial bindings will be generated using gr_modtool bind 
-- ******************************************************************
```

michael-roe, 2021-11-25:
Update: Maybe this isn't as big a problem as I first thought - it looks like most of gnuradio will build without it.

dkozel, 2021-11-25:
Unless you're using gr_modtool there is no problem. If you want to upgrade the gccxml version you could do a pip install into your user account and add to your PYTHONPATH. I believe that can be done without removing the system installed version.

That said, I don't think I had anything else using the system version. If that's true you could uninstall the system version and use pip to get the newer version.

michael-roe, 2021-11-26:
Hi Derek,

Thanks for the reply. 

Unfortunately, I will be using gr_modtool, as I'm writing new blocks. So I may have to solve this dependency problem.

mormj, 2021-11-29:
pybind11 2.x can be installed with pip if you first uninstall apt remove python3-pygccxml

That being said, we probably want to roll back the min requirement for pygccxml to match Ubuntu 20.04 - this was mentioned earlier by @drmpeg  - since we rolled back the std::any features that were causing it to be required

mormj, 2022-01-20:
Closing this - we left the requirement as 2.0.0 because it is readily available and if you put anything c++17 in any header that is transitively included, pygccxml < 2 will choke.

closed 2022-01-20 by mormj