gnuradio/gnuradio #5353: Transmitting is finished Sooner than promised, How I can compute the time of transmitting for a file that is set to no-repeat

stackprogramer, 2021-11-25:
### What happened?

I created a file complex with a file sink in Gnuradio. It is a cos wave.
The size of a recorded file is 1,146,283,728 bytes.  because data type is complex so recorded file has N samples...

N=1,146,283,728 bytes/(8 bytes)=143285466  samples....
All of the samples in the recorded file are 143,285,466. So in another stage, I want to transmit it with Hackrf.......

[image]

In the transmitter stage, I had a GRC file that you can see a screenshot of it.I set the sample rate to 32k.
According to the definition sample rate we expect to consume 32k samples every second.....
So I want to predict that the second Hackrf should be transmitted and red led the show on...
Second= 143,285,466/32000=44776.708125  second but in transmitting it only takes a long 80 seconds.
My question is How I can compute the time of transmitting for a file that is set to no-repeat...
For the Wav file and float type, I had the same problem...
Thanks in advance

[image]

### System Information

OS: (Linux Ubuntu 20.04)
GR Installation Method: (native distro packages - apt)
3.8.1.0 (Python 3.8.10)

### GNU Radio Version

3.8 (maint-3.8)

### Specific Version

3.8.1.0 (Python 3.8.10)

### Steps to Reproduce the Problem

I shared screenshots and describe the problem.

dkozel, 2021-11-25:
Hi. The HackRF has a minimum sample rate of about 2MS/s, so 32kHz will not work. I see mentions online that sometimes the hardware will work at lower rates like 1MS/s, but out doesn't seem to be officially suported and the driver is probably coercing your value to a higher rate internally. If that is the case and there is no warning message then I suggest filling an issue on the HackRF driver.

I hope that helps, if moving to a rate 2MS/s or higher doesn't solve the problem in GNU Radio can you please post again and we can look into it further!

closed 2021-11-25 by dkozel

marcusmueller, 2021-11-25:
You posted this question 10 minutes earlier to the GNU Radio mailing list.

1. Please don't do such things – it's the same people on both platforms that have to now read this twice, and as @dkozel did, react to it. You're actively taking time away from our community by just copy and pasting the same thing into multiple places, which hurts the time we could spend on helping people or improving GNU Radio!
2. This is a bug tracker for GNU Radio itself; you're very welcome to report misbehaviour here, but please make sure that it's actually due to GNU Radio or a component of GNU Radio itself misbehaving.