gnuradio/gnuradio #5354: Renaming a module name with gr_modtool

MobiZaman, 2021-11-26:
### Feature Description

Currently, there exists a command for renaming a block in a GNU OOT module, but I could not find any references on how to change the name of a module. Can this feature please be added if possible?

### Feature Urgency

medium (would be nice to have in the near future)

dkozel, 2021-11-26:
Hello Mubashira, Currently modtool is only able to rename individual blocks, not an entire module. It would be a nice enhancement and I think most of the pieces exist to add module renaming for the blocks, but example flowgraphs and any custom utils would have to be manually changed.

The code implementing the rename functionality is in this file:
https://github.com/gnuradio/gnuradio/blob/master/gr-utils/modtool/core/rename.py

I doubt that this feature will be implemented before GNU Radio 3.10 is released (around the end of January), but if someone does want to work on it that would be welcome.

In the meantime manually renaming is possible. I don't have the time at the moment to do a full test, but I think the following commands do at least 90% of the work, possibly everything. **Make sure to have a backup of the entire OOT folder, or do the commands in a copy of the folder to test first!**

```
# To rename gr-oldname to gr-newname
# Run the following commmands from the base directory of the OOT module (ie ~/src/gr-oldname)
find . -type f | xargs -0 sed -i '' -e 's/oldname/newname/g'
find . -type f | xargs -0 sed -i '' -e 's/OLDNAME/NEWNAME/g'

# Then rebind each c++ block
ls -1 include/newname/*.h
# run for each block listed, except api.h
gr_modtool bind blockname
gr_modtool bind nextblockname
...
```

krono-i2, 2024-02-09:
Is this procedure stil valid?
Thanks for help.