gnuradio/gnuradio #5355: Choser in QT GUI Btn/Toggle/CheckBox gnuradio 3.10 is not dynamic

Phil67ago, 2021-11-26:
### What happened?

A chooser is not dynamic, it is static and shift between it's initialized values.
Two 'QT GUI Range' and one 'QT GUI Toggle Button'
Change a value in 'QT GUI Range' and is not transferred thru the toggle button.
You can only flip the initialized values.
Essentials from 'py' file
`
  #if real == bool:      (BUG unknown 'real')
  #self._cfreq_choices = {'Pressed': bool(cfreq2), 'Released': bool(cfreq1)}
  #elif real == str:
  #self._cfreq_choices = {'Pressed': "cfreq2".replace("'",""), 'Released': "cfreq1".replace("'","")}
  #else:
   self._cfreq_choices = {'Pressed': cfreq2, 'Released': cfreq1}
  _cfreq_toggle_button = qtgui.ToggleButton(self.set_cfreq, 'Fx2', self._cfreq_choices, False,"'value'".replace("'",""))

self._cfreq2_win = RangeWidget(self._cfreq2_range, self.set_cfreq2, "Fx2 (MHz)", "counter_slider", float, QtCore.Qt.Horizontal)
self._cfreq1_win = RangeWidget(self._cfreq1_range, self.set_cfreq1, "Fx1 (MHz)", "counter_slider", float, QtCore.Qt.Horizontal)
`
This is what would be preferred to work:
cfreq2/cfreq1 -> _cfreq_choices -> self.set_cfreq

### System Information

OS: (Linux Distro, Mac, Windows)
GR Installation Method: (native distro packages - apt, rpm, etc., Conda, Source
pybombs build

### GNU Radio Version

3.10-git (master)

### Specific Version

latest

### Steps to Reproduce the Problem

Always

willcode, 2021-11-26:
It sounds like you would like multiple GUI elements to be synchronized somehow? I'm not sure what you are asking for regarding a "dynamic" toggle button.

dkozel, 2021-11-26:
The feature request is to add a callback to the button values and labels fields such that another block could change the options/values of the chooser dynamically at runtime. (as far as I understand it)

Phil67ago, 2021-11-26:
willcode: I have two Range values and one button to select which one I want my value from.

dkozel: Sounds good!!

willcode, 2021-11-26:
So the feature would be:
- Values of a Toggle (currently set at init and held in `pressReleasedDict`) can be set at runtime
- Values can be variables representing the state of other widgets, e.g., Range
- When the "other widget" changes, GRC arranges to have the Toggle notified (this may be close to how it works currently)
- A change in the active Toggle value sets the associated variable and fires the associated callback

Phil67ago, 2021-11-27:
Ok, Good!

Phil67ago, 2022-12-06:
A patch fixing this...
[gnuradio-toggle-switch-buttons-3.11.patch.txt](https://github.com/gnuradio/gnuradio/files/10166243/gnuradio-toggle-switch-buttons-3.11.patch.txt)

Can also be fixed in installed gnuradio:  share/gnuradio/grc/blocks/qtgui_togglebutton.block.yml

Phil67ago, 2022-12-08:
Ok! 
It seems to be fixed already. https://github.com/gnuradio/gnuradio/pull/6316
In the future, when it comes to dynamic passing thru a value, should that code
also be submitted as a PR?

willcode, 2022-12-08:
Any proposed changes should be submitted as a PR. Go ahead and close this issue if it is fixed by #6316.