gnuradio/gnuradio #5356: Python bindings for logger.h are out of sync

frspin, 2021-11-28:
Compiling Gnuradio from GitHub on Ubuntu 20.04 - 64 bit

This is error message:

CMake Error at cmake/Modules/GrPybind.cmake:86 (message):
  Python bindings for logger.h are out of sync
Call Stack (most recent call first):
  gnuradio-runtime/python/gnuradio/gr/bindings/CMakeLists.txt:86 (GR_PYBIND_MAKE_CHECK_HASH)

How can I solve this error?

Regards
Franco Spinelli
IW2DHW

mormj, 2021-11-28:
It is probably trying to compile against your installed headers rather than those in your source tree.

Delete the logger.h header from your installation of gnu radio, then try re-compiling.  This is a known issue with our cmake.

frspin, 2021-11-28:
Il 28/11/21 12:12, mormj ha scritto:
> It is probably trying to compile against your installed headers rather
> than those in your source tree.
> 
> Delete the logger.h header from your installation of gnu radio, then try
> re-compiling. This is a known issue with our cmake.

Removing logger.h installed by gnuradio
(/usr/local/include/gnuradio/logger.h) don't solve the problem

Removing spdlog .dev package give a "missing spdlog" error.

-- Could NOT find spdlog (missing: spdlog_DIR)
-- 
-- Configuring gnuradio-runtime support...
--   Dependency Boost_FOUND = TRUE
--   Dependency PYTHONINTERP_FOUND = TRUE
--   Dependency MPLIB_FOUND = TRUE
--   Dependency spdlog_FOUND = 0
--   Dependency pybind11_FOUND = 1
CMake Error at cmake/Modules/GrComponent.cmake:63 (message):
  user force-enabled gnuradio-runtime but configuration checked failed
Call Stack (most recent call first):
  gnuradio-runtime/CMakeLists.txt:31 (GR_REGISTER_COMPONENT)

Regards
Franco Spinelli
IW2DHW

frspin, 2021-11-28:
For some strange reason my copy of gnuradio-runtime/include/gnuradio/logger.h was not updated by "git pull" command and was different from file on GitHub.
Updated "by hand" and now problem is gone.

Franco Spinelli
IW2DHW

mormj, 2021-11-28:
That's very strange.  Closing this as a GR issue since it seems to have been resolved with a git pull.

closed 2021-11-28 by mormj

swethakare, 2023-11-04:
I have a similar issue but in burt_tagger.h how will i resolve this?

paul-k-young, 2024-01-28:
I had a similar problem when trying to compile an OOT module from Windows.  It turns out that I had to disable the git setting that converts the line termination characters.

git config --global core.autocrlf false

I then deleted the headers and reverted the files to force git to restore the unmodified files.

eabase, 2024-11-06:
I have the same issue with: 
`Python bindings for channelizer.h are out of sync`
...and I already have the `core.autocrlf` set to false.

marcusmueller, 2024-11-06:
@eabase  which version of GNU Radio's source code are you on, *exactly*? (`git describe` has the output to describe that).

eabase, 2024-11-08:
Hi Marcus, 
> which version of GNU Radio's source code are you on, exactly?

No idea, since I'm not using source install. I'm using the MSYS / MINGW64 package. 
Maybe: `3.10.11.0-6` ?
(I now see there is a later version.)

https://packages.msys2.org/base/mingw-w64-gnuradio
https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-gnuradio/PKGBUILD

```bash
# pacman -Sii mingw64/mingw-w64-x86_64-gnuradio

Repository      : mingw64
Name            : mingw-w64-x86_64-gnuradio
Version         : 3.10.11.0-6
Description     : General purpose DSP and SDR toolkit (mingw-w64)
Architecture    : any
URL             : https://gnuradio.org/
Licenses        : spdx:GPL-3.0-or-later
Groups          : None
Provides        : None
Depends On      : mingw-w64-x86_64-codec2  mingw-w64-x86_64-boost  mingw-w64-x86_64-gcc-libs  mingw-w64-x86_64-gmp  mingw-w64-x86_64-gsl  mingw-w64-x86_64-gsm
                  mingw-w64-x86_64-gtk3  mingw-w64-x86_64-libad9361-iio  mingw-w64-x86_64-libiio  mingw-w64-x86_64-libuhd  mingw-w64-x86_64-libunwind
                  mingw-w64-x86_64-fftw  mingw-w64-x86_64-mathjax2  mingw-w64-x86_64-volk  mingw-w64-x86_64-python-click  mingw-w64-x86_64-python-click-plugins
                  mingw-w64-x86_64-python-gobject  mingw-w64-x86_64-python-cairo  mingw-w64-x86_64-python-jsonschema  mingw-w64-x86_64-python-mako
                  mingw-w64-x86_64-python-matplotlib  mingw-w64-x86_64-python-numpy  mingw-w64-x86_64-python-pygccxml  mingw-w64-x86_64-python-pyaml
                  mingw-w64-x86_64-python-pyzmq  mingw-w64-x86_64-python-scipy  mingw-w64-x86_64-python-qtpy  mingw-w64-x86_64-python-pyqt5
                  mingw-w64-x86_64-python-pyqtgraph  mingw-w64-x86_64-python-lxml  mingw-w64-x86_64-python-pyopengl  mingw-w64-x86_64-qt5-base
                  mingw-w64-x86_64-angleproject  mingw-w64-x86_64-qwt-qt5  mingw-w64-x86_64-SDL  mingw-w64-x86_64-soapysdr  mingw-w64-x86_64-portaudio
                  mingw-w64-x86_64-libsndfile  mingw-w64-x86_64-spdlog  mingw-w64-x86_64-fmt  mingw-w64-x86_64-zeromq
Optional Deps   : mingw-w64-x86_64-soapyrtlsdr: install to use RTL-SDR devices with gnuradio
Required By     : None
Optional For    : None
Conflicts With  : None
Replaces        : None
Download Size   : 14.91 MiB
Installed Size  : 326.85 MiB
Packager        : CI (msys2/msys2-autobuild/22f1e5ad/10859909017)
Build Date      : Sat Sep 14 07:38:16 2024
MD5 Sum         : None
SHA-256 Sum     : a6cd18ab66789844cea1ab1d76ea7356226e240d6705209aa970ab82c57ac2dc
Signatures      : None
Extended Data   : None
```