gnuradio/gnuradio #5364: gr_modtool bind not working on 3.10

mormj, 2021-12-02:
### What happened?

getting pygccxml error when using `gr_modtool bind` with an OOT created with GR 3.10

### System Information

OS: Ubuntu 20.04
GR Installation Method: source

### GNU Radio Version

3.10-git (master)

### Specific Version

3.10.0.0-rc1

### Steps to Reproduce the Problem

gr_modtool newmod asdf
cd gr-asdf
gr_modtool add -t sync -l cpp foo
gr_modtool bind foo

### Relevant log output

```shell
(('/usr/include/c++/9/type_traits', 599), ('::', 'std', 'type'))
```

willcode, 2021-12-02:
This works, compiles, runs on F35.

gcc version 11.2.1 20210728 (Red Hat 11.2.1-1) (GCC)

mormj, 2021-12-02:
ok, maybe my prefix or pygccxml version is screwed.

What do you get for 
`python3 -c "import pygccxml; print(pygccxml.__version__)"`

willcode, 2021-12-02:
Interesting, pygccxml wasn't installed. When I installed it, castxml was missing. When I installed that I got
```
$ gr_modtool bind foo

GNU Radio module name identified: asdf
INFO Parsing source file "/home/jwl/grmaster/gr-asdf/include/gnuradio/asdf/foo.h" ... 
In file included from /home/jwl/grmaster/gr-asdf/include/gnuradio/asdf/foo.h:12:
In file included from /home/jwl/grmaster/include/gnuradio/sync_block.h:15:
In file included from /home/jwl/grmaster/include/gnuradio/block.h:17:
In file included from /home/jwl/grmaster/include/gnuradio/basic_block.h:15:
In file included from /home/jwl/grmaster/include/gnuradio/io_signature.h:17:
In file included from /home/jwl/grmaster/include/gnuradio/buffer_double_mapped.h:15:
In file included from /home/jwl/grmaster/include/gnuradio/buffer.h:15:
In file included from /home/jwl/grmaster/include/gnuradio/custom_lock.h:15:
In file included from /home/jwl/grmaster/include/gnuradio/logger.h:27:
In file included from /usr/include/spdlog/common.h:36:
/usr/include/spdlog/fmt/fmt.h:22:14: fatal error: 'spdlog/fmt/bundled/core.h' file not found
#    include <spdlog/fmt/bundled/core.h>
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
Error occurred while running CASTXML:  status:1
```

willcode, 2021-12-02:
pygccxml 2.2.1 installed via pip, castxml 0.4.4 via dnf.
spdlog-devel-1.9.2-2.fc35.x86_64

willcode, 2021-12-02:
This is some sort of spdlog packaging problem. Based on some random posts, I added this line to `include/gnuradio/logger.h`:

#define SPDLOG_FMT_EXTERNAL

Then, after what seemed like too many seconds, I got:

`(('/usr/include/c++/11/type_traits', 530), ('::', 'std', 'type'))`

mormj, 2021-12-02:
It's happening deep inside pygccxml - castxml runs fine but when it starts manipulating the xml data, things go haywire
But the same version of pygccxml and castxml with a module created with 3.9 works perfectly normal.  Something in the restructuring of the oot

mormj, 2021-12-02:
I think it has nothing to do with the restructuring of the OOT, but instead something in the base block class structure.  This might be related to the custom buffers

If I comment out #include <gnuradio/sync_block.h> and then don't make the block inherit from sync_block, pygccxml will run

so this is a 3.10 thing that obviously needs to be sorted out

mormj, 2021-12-02:
I have a bad feeling the problem is the introduction of this:
```
#include <gnuradio/buffer_type.h>
```

That comes in through io_signature.h

mormj, 2021-12-03:
Nope, that is not the problem - looks like the problem is `logging.h` and some of the spdlog includes

Can be reproduced by creating an OOT, adding a block.  Now remove the inheritance from sync_block and comment out #include<sync_block.h>  -- this should bind fine

Start adding includes back in --> logging.h is where it breaks, and specifically at the spdlog/fmt includes in there

PR #5366 (2021-12-06): Modtool bind with spdlog complication

## Description
after the changes introduced in logger.h, pygccxml does not play well with the spdlog
headers which are implicitly included in every block.  This caused `gr_modtool bind`
to crash 

## Related Issue
Fixes #5364 

## Which blocks/areas does this affect?
Any OOT using gr_modtool bind

## Testing Done
Created a new OOT with a few blocks and then ran `gr_modtool bind`

## Checklist

- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/master/CONTRIBUTING.md).
- [x] I have squashed my commits to have one signif [... cut]
closed 2021-12-06 by mormj