gnuradio/gnuradio #5368: examples/filter/filter_taps.grc

cinaeds, 2021-12-04:
### What happened?

Exited with error. 

File "./filter_taps.py", line 172
    self.set_br_taps(firdes.<class 'type'>(

SyntaxError: invalid syntax

### System Information

Debian 10 buster - GR was built from source code.

### GNU Radio Version

3.8 (maint-3.8)

### Specific Version

3.8.40

### Steps to Reproduce the Problem

Simple run the ./examples/filter/filter_taps.grc code shipped with the source code. 

To fix the problem I replaced 

   self.set_br_taps(firdes.<class 'type'>(...) 

with 

  self.set_br_taps(firdes.band_reject(...)

willcode, 2021-12-04:
Thanks for the report. There was a bug in the grc yml file for that block.

willcode, 2021-12-04:
Fixed on maint-3.8. Versions 3.9+ were already ok.

PR #5370 (2021-12-04): fix typo in band reject taps variable block set_taps() (maint-3.8)
# Pull Request Details

## Description

GRC YML filter for band reject filter variable contained a typo in the set_taps() callback. Resulting generated code cause did not run.

Callback was added in b405ca8, but using the logic for the band pass taps block which has multiple types. Band reject only supports float.

## Related Issue

Fixes #5368

## Which blocks/areas does this affect?

Band-reject Filter Taps

## Testing Done
closed 2021-12-04 by willcode