gnuradio/gnuradio #5373: module 'lora' has no attribute 'lora_receiver'

silos96, 2021-12-07:
### What happened?

[image]
[image]

### System Information

OS: (Linux Distro, Mac, Windows)
GR Installation Method: (native distro packages - apt, rpm, etc., Conda, Source

### GNU Radio Version

3.8 (maint-3.8)

### Specific Version

3.8.3.1

### Steps to Reproduce the Problem

the problem appears when i execute the programs

### Relevant log output

```shell
Generating: '/home/francesco/Scrivania/qam_test/QAM_mod.py'
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: /usr/bin/python3 -u /home/francesco/Scrivania/qam_test/QAM_mod.py

Traceback (most recent call last):
  File "/home/francesco/Scrivania/qam_test/QAM_mod.py", line 290, in <module>
    main()
  File "/home/francesco/Scrivania/qam_test/QAM_mod.py", line 266, in main
    tb = top_block_cls()
  File "/home/francesco/Scrivania/qam_test/QAM_mod.py", line 216, in __init__
    self.digital_qam_mod_0 = digital.qam.qam_mod(
AttributeError: module 'gnuradio.digital.qam' has no attribute 'qam_mod'

>>> Done (return code 1)
```

silos96, 2021-12-07:
i use ubuntu 20.04.3 lts

willcode, 2021-12-07:
The QAM Mod and QAM Demod blocks were intended to be deleted in 9fa00984d8 back in 2018. Only the code was deleted, though, and the GRC YAML files were left in. Solution is to delete the blocks so users do not think they exist.

Instead, the Constellation Modulator and Constellation Demodulator should be used, along with a Constellation Object representing QAM.

willcode, 2021-12-07:
@silos96 If you are also having trouble with a LoRa OOT, please post an issue on that project's site.

silos96, 2021-12-07:
Many many thanks for your answer, I solve the problem with the block that
you has indicated.

——————————————————————
Francesco Silino, Ph.D in Electronic Engineering
Telecommunications & Remote Sensing Laboratory
Dept. Of Electrical, Computer and Biomedical Engineering
University of Pavia
Via Ferrata 5, 27100 Pavia, Italy
MAT. 509722

closed 2021-12-07 by willcode