gnuradio/gnuradio #5379: Example out of tree module won't build

michael-roe, 2021-12-08:
### What happened?

If I try to build the example out-of-tree module from the tutorial, cmake fails because $(SRC)/gr-howto/python/howto/bindings/square2_ff_python.cc doesn't exist.

(I used pip to install the most recent version of pygccxml rather than using the too old version from the Ubuntu distribution).

### System Information

OS: Ubuntu 20.04
GR Installation Method: source

### GNU Radio Version

3.10-git (master)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

gr_modtool newmod  howto
cd gr-howto
gr_modtool add -t sync -l cpp square2_ff
cd ~/build/gr-howto
cmake ~/src/gr-howto

### Relevant log output

```shell
mr101@Fieldmouse:~/build/gr-howto$ cmake ~/src/gr-howto
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gmp'
--   Found gmp, version 6.2.0
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Using GMP.
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.71.0") found components: date_time program_options system regex thread unit_test_framework 
-- Found Volk: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.10") 
-- Found Git: /usr/bin/git  
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen missing components: dot
-- Using install prefix: /usr/local
-- Building for version: 1.0.0.0 / 1.0.0
-- No C++ unit tests... skipping
-- PYTHON and GRC components are enabled
-- Python checking for pygccxml - found
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so
Traceback (most recent call last):
  File "/home/mr101/src/gr-howto/python/howto/bindings/header_utils.py", line 69, in <module>
    pbhp = PybindHeaderParser(args.pathname)
  File "/home/mr101/src/gr-howto/python/howto/bindings/header_utils.py", line 9, in __init__
    with open(pathname, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/mr101/src/gr-howto/python/howto/bindings/square2_ff_python.cc'
CMake Error at /usr/local/lib/cmake/gnuradio/GrPybind.cmake:199 (list):
  list GET given empty list
Call Stack (most recent call first):
  python/howto/bindings/CMakeLists.txt:34 (GR_PYBIND_MAKE_OOT)

CMake Error at /usr/local/lib/cmake/gnuradio/GrPybind.cmake:200 (list):
  list GET given empty list
Call Stack (most recent call first):
  python/howto/bindings/CMakeLists.txt:34 (GR_PYBIND_MAKE_OOT)

CMake Error at /usr/local/lib/cmake/gnuradio/GrPybind.cmake:201 (list):
  list GET given empty list
Call Stack (most recent call first):
  python/howto/bindings/CMakeLists.txt:34 (GR_PYBIND_MAKE_OOT)

CMake Error at /usr/local/lib/cmake/gnuradio/GrPybind.cmake:202 (list):
  list GET given empty list
Call Stack (most recent call first):
  python/howto/bindings/CMakeLists.txt:34 (GR_PYBIND_MAKE_OOT)

-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- LTO enabled
-- Configuring incomplete, errors occurred!
See also "/home/mr101/build/gr-howto/CMakeFiles/CMakeOutput.log".
See also "/home/mr101/build/gr-howto/CMakeFiles/CMakeError.log".
```

dkozel, 2021-12-08:
With the reorg of new OOT modules the tutorial at https://wiki.gnuradio.org/index.php/OutOfTreeModules#Tutorial_3:_Writing_a_signal_processing_block_in_Python is now out of date for the development code that will become 3.10.

However your exact problem looks like an issue with the gr_modtool templates.

willcode, 2021-12-08:
It looks like you are either missing a dependency (e.g., PyBind11) or you do not have PYTHONPATH or LD_LIBRARY_CONFIG set up. I verified that the following works:
```
gr_modtool newmod howto
cd gr-howto
gr_modtool add -t sync -l cpp square2_ff
mkdir build
cd build/
cmake -DCMAKE_INSTALL_PREFIX=/path/to/my/gr3.10/installation ..
make
make test
```
Could be a doc problem, as @dkozel mentioned.

michael-roe, 2021-12-08:
I had previously built and installed gr-howto on the same computer, using a previous version of gnuradio.

What I think may be happening here is that an out of date installed copy of gr-howto is getting picked up rather than the new one that is being built.

willcode, 2021-12-08:
Possible. Also, 3.10 has had some recent changes (it is not yet a stable release) that could cause this. If you figure it out, please let us know what the problem was and whether you found any documentation to be wrong. You can close this PR if you are all set.

duggabe, 2023-07-15:
the reference page has been deprecated. the replacement page works. this should be closed.

closed 2023-07-15 by willcode