gnuradio/gnuradio #5387: argument

Lijinyudoudou, 2021-12-10:
### What happened?

When I tried to write a  embedded python Block, and  it works but 
this error occours.
<pre>handler caught exception: in method &apos;block_gw_message_type_work_args_return_value_set&apos;, argument 2 of type &apos;int&apos;
Traceback (most recent call last):
  File &quot;/usr/lib/python3/dist-packages/gnuradio/gr/gateway.py&quot;, line 60, in eval
    try: self._callback()
  File &quot;/usr/lib/python3/dist-packages/gnuradio/gr/gateway.py&quot;, line 218, in __gr_block_handle
    self.__message.work_args_return_value = self.work(
TypeError: in method &apos;block_gw_message_type_work_args_return_value_set&apos;, argument 2 of type &apos;int&apos;
thread[thread-per-block[2]: &lt;block Embedded Python Block(3)&gt;]: SWIG director method error. Error detected when calling &apos;feval_ll.eval&apos;
handler caught exception: index 6 is out of bounds for axis 0 with size 6
Traceback (most recent call last):
  File &quot;/usr/lib/python3/dist-packages/gnuradio/gr/gateway.py&quot;, line 60, in eval
    try: self._callback()
  File &quot;/usr/lib/python3/dist-packages/gnuradio/gr/gateway.py&quot;, line 218, in __gr_block_handle
    self.__message.work_args_return_value = self.work(
  File &quot;/home/jinyulee/Desktop/epy_block_2.py&quot;, line 33, in work
    print(input_items[0][(self.sf-i)-1] )
IndexError: index 6 is out of bounds for axis 0 with size 6
thread[thread-per-block[1]: &lt;block decimal(2)&gt;]: SWIG director method error. Error detected when calling &apos;feval_ll.eval&apos;
</pre>

It's happend due to wrong output or return i set ?

and Following is the code for Block
import numpy as np
from gnuradio import gr

class blk(gr.sync_block):  # other base classes are basic_block, decim_block, interp_block
    """Embedded Python Block example - a simple multiply const"""

    def __init__(self, sf =7 ):  # only default arguments here
        """arguments to this function show up as parameters in GRC"""
        gr.sync_block.__init__(
            self,
            name='decimal',   # will show up in GRC
            in_sig=[np.byte],
            out_sig=[np.int16]
        )
        # if an attribute with the same name as a parameter is found,
        # a callback is registered (properties work, too).
        self.sf =sf 

    def work(self, input_items, output_items):
        #print(input_items[0])
        n = len(output_items[0])
        result = 0
        for i in range(n):
            print(input_items[0][(self.sf-i)-1] ) 
            result = input_items[0][(self.sf-i)-1] * (2**i) + result
        print("Binary to Decical data")
        print(result)
        #output_items[0] = result
            #print(input_items[0])    
        return len(output_items)

### System Information

OS: (Linux Ubuntu20.4)
GR Installation Method: (native distro packages - apt)

### GNU Radio Version

3.8 (maint-3.8)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

When I tried to write a  embedded python Block, and  it works but 
this error occours.

### Relevant log output

```shell
<pre>handler caught exception: in method &apos;block_gw_message_type_work_args_return_value_set&apos;, argument 2 of type &apos;int&apos;
Traceback (most recent call last):
  File &quot;/usr/lib/python3/dist-packages/gnuradio/gr/gateway.py&quot;, line 60, in eval
    try: self._callback()
  File &quot;/usr/lib/python3/dist-packages/gnuradio/gr/gateway.py&quot;, line 218, in __gr_block_handle
    self.__message.work_args_return_value = self.work(
TypeError: in method &apos;block_gw_message_type_work_args_return_value_set&apos;, argument 2 of type &apos;int&apos;
thread[thread-per-block[2]: &lt;block Embedded Python Block(3)&gt;]: SWIG director method error. Error detected when calling &apos;feval_ll.eval&apos;
handler caught exception: index 6 is out of bounds for axis 0 with size 6
Traceback (most recent call last):
  File &quot;/usr/lib/python3/dist-packages/gnuradio/gr/gateway.py&quot;, line 60, in eval
    try: self._callback()
  File &quot;/usr/lib/python3/dist-packages/gnuradio/gr/gateway.py&quot;, line 218, in __gr_block_handle
    self.__message.work_args_return_value = self.work(
  File &quot;/home/jinyulee/Desktop/epy_block_2.py&quot;, line 33, in work
    print(input_items[0][(self.sf-i)-1] )
IndexError: index 6 is out of bounds for axis 0 with size 6
thread[thread-per-block[1]: &lt;block decimal(2)&gt;]: SWIG director method error. Error detected when calling &apos;feval_ll.eval&apos;
</pre>
```

willcode, 2021-12-13:
Likely an error in your work() function. The issue tracker is for feature requests and bug reports. Please ask for help on chat.gnuradio.org or the mailing list.

closed 2021-12-13 by willcode

dkozel, 2021-12-13:
Hi @Lijinyudoudou The best place to ask questions about code you've written is in the chat channel.
https://app.element.io/#/room/#gnuradio:gnuradio.org

If you are converting a series of binary bits where each bit is in one byte item, check the Unpacked to Packed block.
https://wiki.gnuradio.org/index.php/Packed_to_Unpacked

Something you'd encounter as well is that the block will only produce 1 output item for each N input bits (sf in your code?) that you're packing together.