gnuradio/gnuradio #5398: gr_filter_design tracebacks: AttributeError: module 'PyQt5.QtGui' has no attribute 'QGraphicsObject'

yarda, 2021-12-15:
### What happened?

```
$ /usr/bin/gr_filter_design
Traceback (most recent call last):
  File "/usr/bin/gr_filter_design", line 11, in <module>
    from gnuradio.filter import filter_design
  File "/usr/lib64/python3.10/site-packages/gnuradio/filter/filter_design.py", line 56, in <module>
    from gnuradio.filter.banditems import *
  File "/usr/lib64/python3.10/site-packages/gnuradio/filter/banditems.py", line 18, in <module>
    class filtermovlineItem(QtGui.QGraphicsObject):
AttributeError: module 'PyQt5.QtGui' has no attribute 'QGraphicsObject'
```

### System Information

OS: (Fedora 35, rawhide/36)
GR Installation Method: (native distro package)

### GNU Radio Version

3.9 (maint-3.9)

### Specific Version

3.9.4.0

### Steps to Reproduce the Problem

`$ /usr/bin/gr_filter_design`

### Relevant log output

```shell
Traceback (most recent call last):
  File "/usr/bin/gr_filter_design", line 11, in <module>
    from gnuradio.filter import filter_design
  File "/usr/lib64/python3.10/site-packages/gnuradio/filter/filter_design.py", line 56, in <module>
    from gnuradio.filter.banditems import *
  File "/usr/lib64/python3.10/site-packages/gnuradio/filter/banditems.py", line 18, in <module>
    class filtermovlineItem(QtGui.QGraphicsObject):
AttributeError: module 'PyQt5.QtGui' has no attribute 'QGraphicsObject'
```
```

yarda, 2021-12-15:
It seems the QGraphicsObject is in the QtWidgets not in the QtGui. Fedora downstream bugzilla:
https://bugzilla.redhat.com/show_bug.cgi?id=2032668

willcode, 2021-12-15:
This has already been fixed on master and backported to maint-3.9, so it will work on the next releases. See #5276 and #5342.

closed 2021-12-15 by willcode